RT-Thread/rt-thread #4140: 这里头文件包含还有问题。

liuduanfei, 2020-12-09:
https://github.com/RT-Thread/rt-thread/blob/4f7cff88d08f0a2de87573a577dcd25b6e7f5544/components/drivers/include/drivers/audio.h#L16

audio.h包含了audio_pipe.h，audio_pipe.h里面包含了rtdevice.h，rtdevice.h又包含了audio.h，导致编译得时候出现：

audio.h:150:26: error: field 'pipe' has incomplete type struct rt_audio_pipe pipe;

thread-liu, 2020-12-10:
根据 PR ：https://github.com/RT-Thread/rt-thread/pull/4055  的修改， IAR 和 Studio 编译都会出问题：  
IAR:  
[image]
Studio:  
[image]

liuduanfei, 2020-12-10:
> 根据 PR ：#4055 的修改， IAR 和 Studio 编译都会出问题：
> IAR:
> [image]
> Studio:
> [image]

是的，在audio_pipe.c中加上 #include<rtdevice.h>能解决编译错误的问题，但是这里头文本包含还是别扭着。

thread-liu, 2020-12-10:
在 PR ： https://github.com/RT-Thread/rt-thread/pull/4055/files 的基础上修改 audio_pipe.h :
#include <rtdevice.h> -> #include "ipc/ringbuffer.h"

liuduanfei, 2020-12-10:
> 在 PR ： https://github.com/RT-Thread/rt-thread/pull/4055/files 的基础上修改 audio_pipe.h :
> #include <rtdevice.h> -> #include "ipc/ringbuffer.h"
> 这样 audio_pipe.h 就不会包含 audio.h, 编译也可以通过

嗯嗯，这样也可以。

closed 2020-12-13 by liuduanfei