RT-Thread/rt-thread #4144: K210's global interrupt cannot be disabled completely

bj-wanghz, 2020-12-10:
**Description**

When using Micropython on K210, global interrupt cannot be disabled all the time by `machine.disable_irq()` which actually invokes `rt_hw_interrupt_disable()`. After `machine.disable_irq()` is executed, we expect no input will be accepted by *REPL (Micropython shell)* as it takes user input from UARTHS receiver interrupt routine. But unfortunately, it still works. 

**Analysis**

Assume that Micropython executes in Thread A. There are also other threads in the system, and we take one of them as Thread B. 

Thread switch is done by `rt_schedule()` function, in which global interrupt is operated like the following:

```C
/* src/scheduler.c */

void rt_schedule(void)
{
    ...
    level = rt_hw_interrupt_disable();
    ...
    // switch to another thread
    
    rt_hw_interrupt_enable(level);	// when being switched back
}
```

- If Thread B gets CPU time first and then switch to Thread A, in `rt_schedule()`Thread B will save a value into `level` which indicates global interrupt is enabled. 
- When Thread A starts to run, and after global interrupt is disabled explicitly, for example, by a `machine.disable_irq()` call, switch to Thread B happens. 
- Thread B will continue to execute `rt_hw_interrupt_enable(level)` at the end of `rt_schedule()`. And the problem occurs. Because Thread B has saved an *enabled* status in `level`, the global interrupt is enabled by this call, although Thread A has disabled it. 
- When Thread B runs, UARTHS receive *irq* can still accept characters and save them into the ringbuffer. So REPL can continue to work in Thread A.

**Discuss**

I haven't figured out any proper solution for this problem. Maybe locking the scheduler in `rt_hw_interrupt_disable()` is an option?

BernardXiong, 2020-12-11:
How to use machine.disable_irq()? If you want to disable all of them, you can try it with:

```c
machine.disable_irq()
{
    rt_hw_interrupt_disable();
    rt_exit_critical();
}
```

And, please notice the SMP case.

bj-wanghz, 2020-12-14:
> How to use machine.disable_irq()? If you want to disable all of them, you can try it with:
> 
> ```c
> machine.disable_irq()
> {
>     rt_hw_interrupt_disable();
>     rt_exit_critical();
> }
> ```
> 
> And, please notice the SMP case.

What is expected is to disable all the interrupt requests to the current core by `machine.disable_irq()` until the code enables them again explicitly. But this doesn't work even in a sing-core system, as context switch is not implemented by an ISR for risc-v (also arm9?). While `cortex-M3` doesn't suffer this problem as it switches context by `pendsv`, which can be disabled by `rt_hw_interrupt_disable()`. 
Using `rt_enter_critical()` and `rt_hw_interrupt_disable()` may work, but it seems a little weird because users have to remember they can only disable interrupts by this two statements.

BernardXiong, 2020-12-15:
Yes, my mistaken. It shall be `rt_enter_critical`.

closed 2020-12-29 by bj-wanghz