RT-Thread/rt-thread #4176: k210 使用rw007 lwip协议使用2.1.2版本导致出问题

chunyexixiaoyu, 2020-12-18:
[image]

chunyexixiaoyu, 2020-12-18:
[image]

BernardXiong, 2020-12-18:
这个是初始化就错误了？

chunyexixiaoyu, 2020-12-18:
> 这个是初始化就错误了？
是的，我移植的rw007包现在是可以用了，但是我menuconfig 选择lwip协议版本的时候选择最新的会报上面的错误，然后在设置里面都写了8字节对齐 存储

BernardXiong, 2020-12-18:
> > 这个是初始化就错误了？
> > 是的，我移植的rw007包现在是可以用了，但是我menuconfig 选择lwip协议版本的时候选择最新的会报上面的错误，然后在设置里面都写了8字节对齐 存储

还请给个清晰的说明，

1. 选择了v2.1.2版本lwIP？
2. 然后默认4字节对齐初始化报错？
3. 还是说，设置8字节对齐也报错？

chunyexixiaoyu, 2020-12-18:
> > > 这个是初始化就错误了？
> > > 是的，我移植的rw007包现在是可以用了，但是我menuconfig 选择lwip协议版本的时候选择最新的会报上面的错误，然后在设置里面都写了8字节对齐 存储
> 
> 还请给个清晰的说明，
> 
> 1. 选择了v2.1.2版本lwIP？
> 2. 然后默认4字节对齐初始化报错？
> 3. 还是说，设置8字节对齐也报错？

选择了2.12lwip 然后设置8字节对齐就是上面的现象。选择2.12上一个版本选择8字节对齐没错。k210 64位单片机应该是要选择8字节对齐的吧？

heyuanjie87, 2020-12-18:
2.0.2的lwipopts.h是这样写的
#ifdef RT_LWIP_MEM_ALIGNMENT
#define MEM_ALIGNMENT RT_LWIP_MEM_ALIGNMENT
#else
#define MEM_ALIGNMENT               4
#endif

你复制到2.1.2试试，里面是写死的4

chunyexixiaoyu, 2020-12-18:
> 2.0.2的lwipopts.h是这样写的
> #ifdef RT_LWIP_MEM_ALIGNMENT
> #define MEM_ALIGNMENT RT_LWIP_MEM_ALIGNMENT
> #else
> #define MEM_ALIGNMENT 4
> #endif
> 
> 你复制到2.1.2试试，里面是写死的4

是这个问题，那就是2.1.2代码的bug了 里面没有进行if else 的判断 。我直接把2.1.2里面的宏定义写成8是可以运行的

closed 2020-12-18 by chunyexixiaoyu