RT-Thread/rt-thread #4182: stm32G070RBT complie fail when enable RTC

MilesWangSeven, 2020-12-20:
IDE: RT-Thread Studio
Project: RT-Thread 4.0.2
MCU: STM32G070RBT

I follow this below
/*-------------------------- HAREWARE TIMER CONFIG END --------------------------*/

/*-------------------------- RTC CONFIG BEGIN --------------------------*/

/** if you want to use rtc(hardware) you can use the following instructions.
 *
 * STEP 1, open rtc driver framework(hardware) support in the RT-Thread Settings file
 *
 * STEP 2, define macro related to the rtc
 *                 such as    BSP_USING_ONCHIP_RTC
 *
 * STEP 3, modify your stm32xxxx_hal_config.h file to support rtc peripherals. define macro related to the peripherals
 *                 such as     #define HAL_RTC_MODULE_ENABLED
 *
 */
#define BSP_USING_ONCHIP_RTC

then complie, I get errors.

-------------------------------------------------------------------
finally, I think this code below should add in board.h
#include <drv_common.h>

BernardXiong, 2020-12-22:
Thank you for your feedback. @margguo Please take a look, thanks.

margguo, 2020-12-22:
@MilesWangSeven ,  you are right， this does cause an error while open some peripheral for the miss including of “stm32g0xx_hal_conf.h”, drv_comman.h will also work for there is an including of stm32g0xx_hal_conf.h

Background:  RT-Thread Studio is designed to compatible with various  IC vendors, including ST. And "borad.h" is an IDE generated hardware config file, it would not include any code that only works with STM32.

Workaround: we would fix this issue by update the STM32G0 chip support pack, solving this error by adding a hook, which will be called after the project is created, the hook will add "#include <stm32g0xx.h>#include <drv_common.h>" to board.h

Thanks for your feedback.

closed 2021-03-26 by mysterywolf