RT-Thread/rt-thread #4184: NXP i.MXRT1050/1060  GPIO Interrupt bug in drv_gpio.c driver file

jicheng0622, 2020-12-21:
In drv_gpio.c, The pin_start input of imxrt_isr API in GPIOx_Combined_16_31_IRQHandler should be changed to 16 instead of 15. The GPIO(x, 31) interrupt flag can not be cleared with the original 15, which results into CPU dead in ISR if GPIO(x,31) interrupt is enabled and its event is detected.

[image]

BernardXiong, 2020-12-22:
@Guozhanxin Please take a look.

jicheng0622, 2020-12-31:
hi @willianchanlovegithub，

Have you verify this issue and do we have plan to fix this issue? THanks.

willianchanlovegithub, 2021-01-20:
@jicheng0622 Sure, We will fix it. Thanks for you proposal.

Guozhanxin, 2021-02-05:
@jicheng0622 If you have confirmed and fixed this problem, you can mention PR to fix it.

mysterywolf, 2021-02-05:
@Guozhanxin  I think we should add a bug label to this issue.

mysterywolf, 2022-08-04:
在此处修复：https://github.com/RT-Thread/rt-thread/pull/6220

closed 2022-08-11 by mysterywolf