RT-Thread/rt-thread #4186: STM32F746G-Discovery 在RT-thread下进行浮点运算会报错

DUXin-ece, 2020-12-22:
具体错误：fault: SCB_CFSR_UFSR:0x01 UNDEFINSTR

DUXin-ece, 2020-12-22:
测试代码如下：
[image]

烧录后运行程序结果如下：
[image]

DUXin-ece, 2020-12-22:
测试代码如下：
[image]
烧录后运行结果如下：
[image]

chunyexixiaoyu, 2020-12-22:
楼上是我们实验室的小伙厉害了

BernardXiong, 2020-12-23:
> 楼上是我们实验室的小伙厉害了

😄 为什么厉害了？别人天生厉害吧。这个问题可以看看栈的情况，或者一些编译器的选项配置。

DUXin-ece, 2020-12-23:
> > 楼上是我们实验室的小伙厉害了
> 
>  为什么厉害了？别人天生厉害吧。这个问题可以看看栈的情况，或者一些编译器的选项配置。

老师您好，目前该问题已解决。该问题是由于rtthread在gcc编译平台下和FPU相关的flag设置错误引起的。STM32F74X系列只支持单精度浮点运算，因此`-mfpu`应该设置为`fpv5-sp-d16`。已经提交了Pull Request。

BernardXiong, 2020-12-23:
> > > 楼上是我们实验室的小伙厉害了
> > 
> > 
> > 为什么厉害了？别人天生厉害吧。这个问题可以看看栈的情况，或者一些编译器的选项配置。
> 
> 老师您好，目前该问题已解决。该问题是由于rtthread在gcc编译平台下和FPU相关的flag设置错误引起的。STM32F74X系列只支持单精度浮点运算，因此`-mfpu`应该设置为`fpv5-sp-d16`。已经提交了Pull Request。

Thank you so much. You are the best one! 👍

closed 2020-12-23 by DUXin-ece