RT-Thread/rt-thread #4227: [Enhancement] Add a USB CDC example

M0n7y5, 2021-01-07:
HI, 
can somebody add simple example using USB CDC?

balanceTWK, 2021-01-08:
[image]

```c
 1  #include <rtthread.h>
 2  #include <rtdevice.h>
 3
 4  int main(void)
 5  {
 6      rt_device_t dev = RT_NULL;
 7      char buf[] = "hello rt-thread!\r\n";
 8
 9      dev = rt_device_find("vcom");
10
11      if (dev)
12          rt_device_open(dev, RT_DEVICE_FLAG_RDWR);
13      else
14          return -RT_ERROR;
15
16      while (1)
17      {
18          rt_device_write(dev, 0, buf, rt_strlen(buf));
19          rt_thread_mdelay(500);
20      }
21
22      return RT_EOK;
23  }
```

You can find examples here！

M0n7y5, 2021-01-08:
Waiting for this ( #4229 ) to be resolved

M0n7y5, 2021-01-12:
@balanceTWK ok so its working, but when i send something from PC to MCU it will crash a get this error: 

(rx_fifo != RT_NULL) assertion failed at function:rt_hw_serial_isr, line number:1165
[image]

balanceTWK, 2021-01-13:
> [image]
>
> ```c
>  1  #include <rtthread.h>
>  2  #include <rtdevice.h>
>  3
>  4  int main(void)
>  5  {
>  6      rt_device_t dev = RT_NULL;
>  7      char buf[] = "hello rt-thread!\r\n";
>  8
>  9      dev = [...]

This is just a simple use of "vcom". you can try to use  rt_device_open(dev, RT_DEVICE_FLAG_RDWR | RT_DEVICE_FLAG_INT_RX ); 

[image]

balanceTWK, 2021-01-13:
[image]

There is another quick way to validate "VCOM". There is a package that can quickly switch Finsh/MSH to "VCOM". Use method is as follows:

[image]

The final experimental results are as follows:

[image]

M0n7y5, 2021-01-13:
Ok, so i was missing RX flag. I believe this is correct implementation of receiving data from CDC (or any other device):
```c
char rx[128];
static rt_err_t rx_ind(rt_device_t dev, rt_size_t size)
{
    // Null terminator
    rx[size + 1] = 0;
    rt_device_read(dev, 0, rx, size);
    LOG_D("CDC RX: %s", rx);

    return RT_EOK;
}

int main(void)
{
    rt_device_t dev = RT_NULL;
    char buf[] = "hello rt-thread!\r\n";
    dev = rt_device_find("vcom");
    rt_err_t err = RT_EOK;

    if (dev)
       err = rt_device_open(dev, RT_DEVICE_FLAG_RDWR | RT_DEVICE_FLAG_INT_RX);
    else
        return -RT_ERROR;

    err = rt_device_set_rx_indicate(dev, &rx_ind);

    while (1)
    {
        //rt_device_read(dev , 0, buffer, size)
        //rt_device_write(dev, 0, buf, rt_strlen(buf));
        rt_thread_mdelay(500);
    }
    return RT_EOK;
} 
```
[image]

Also VConsole is really nice package. I can partially understand how to use it by reading methods names, but rest of it is in Chinese, which i cant read. It would be good to add some rule for uploading packages. For example including English Readme along side with Chinese Readme. I believe there is a lot of cool packages  that i, unfortunately, can't discover :(.

Anyway looks like am gonna read rt-thread docs since everything works correctly using rt-thread api.

closed 2021-01-13 by M0n7y5