RT-Thread/rt-thread #4228: STM32F746-disco crash

updesh31, 2021-01-07:
Hello rt-thread.
 I am using STM32F746-disco for my demo application. I am using PPP,lvgl but after some time system crashes. I am sending you terminal log. Can you help?
psr: 0x01000000
r00: 0x00000000
r01: 0x20041d7c
r02: 0x00001500
r03: 0x720188c5
r04: 0xdeadbeef
r05: 0xdeadbeef
r06: 0xdeadbeef
r07: 0x200405a8
r08: 0xdeadbeef
r09: 0xdeadbeef
r10: 0xdeadbeef
r11: 0xdeadbeef
r12: 0x00000000
 lr: 0x08002605
 pc: 0x0808fbe4
hard fault on thread: FTP

thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
mainapp   20  suspend 0x00000180 0x00001400    27%   0x00000005 000
FTP       30  running 0x00000170 0x0000c800    01%   0x00000009 000
touch     27  suspend 0x000000ac 0x00000800    09%   0x00000012 000
lv-task   20  suspend 0x000000b8 0x00005000    08%   0x0000000a 000
lv_tick   19  suspend 0x000000a8 0x00000200    32%   0x0000000a 000
ppp_recv  21  suspend 0x000000e0 0x00000800    33%   0x00000003 000
tshell    20  suspend 0x00000184 0x00001000    09%   0x00000009 000
usbh       8  suspend 0x000000bc 0x00001000    04%   0x00000014 000
mmcsd_de  22  suspend 0x000000b8 0x00000400    62%   0x0000000a 000
sys_work  23  suspend 0x00000088 0x00000800    26%   0x0000000a 000
tcpip     20  suspend 0x000000f0 0x00002000    20%   0x0000000d 000
etx       22  suspend 0x000000ac 0x00000800    08%   0x00000010 000
erx       22  suspend 0x000000b4 0x00000800    08%   0x00000010 000
tidle0    31  ready   0x0000005c 0x00000400    17%   0x0000000a 000
timer      4  suspend 0x0000007c 0x00000200    33%   0x00000009 000
bus fault:
SCB_CFSR_BFSR:0x82 PRECISERR SCB->BFAR:720188C5

updesh31, 2021-01-07:
On Debug window of RT-Thread Studio I am getting:
<signal handler called>() at 0xfffffffd
rt_hw_interrupt_thread_switch() at context_gcc.S

[image]

BernardXiong, 2021-01-07:
It seems a issue in FTP. If you don't use FTP, is it OK?

updesh31, 2021-01-07:
I got the solution.
First program was:

1 host = gethostbyname(url);
2 server_addr.sin_family = AF_INET;
3 server_addr.sin_port = htons(port);
4 server_addr.sin_addr = *((struct in_addr *)host->h_addr);
5 rt_thread_mdelay(10);
6 rt_memset(&(server_addr.sin_zero), 0, sizeof(server_addr.sin_zero));
7 rt_thread_mdelay(100);
8 ret = connect(m_socket_cmd, (struct sockaddr *)&server_addr, sizeof(struct sockaddr));
In case if host is NULL system crashes at line 4.

So only after checking host NULL condition my problem resolved. Rt-thread working perfectly. :)
Thanks

closed 2021-01-07 by updesh31