RT-Thread/rt-thread #4229: [BUG] SystemClock_Config is missing

M0n7y5, 2021-01-07:
Hi,
i am trying to follow these steps in `board.h`, but it looks like `SystemClock_Config` from `board.c` is missing... 
I am using rt-thread v4.0.2 Standard

## `board.h`
[image]

## `board.c`
[image]

BernardXiong, 2021-01-07:
What's the BSP of this board.c?

M0n7y5, 2021-01-08:
I am using bluepill board with STM32F103CB

M0n7y5, 2021-01-08:
Also i am using rt-thread IDE

BernardXiong, 2021-01-08:
@mysterywolf Do you have some comments?

M0n7y5, 2021-01-08:
I found that method in `Project -> drivers -> drv_clk.c` and i replaced that with my code. Now USB CDC will show up in Device Manager, but still not printing anything. I am using code from here #4227.

M0n7y5, 2021-01-08:
Ok so i managed to get it CDC working ...
[image]

But there is a new problem here. finsh shell is not working anymore for some reason.

margguo, 2021-01-09:
It looks like a prtoject generated from RT-Thread Studio  by chip, not from bsp， replace code in drv_clk.c --system_clock_config with code  from CubeMX fucntion ”SystemClock_Config“ , and pay attention to update the Marco value of HSE_VALUE in stm32fxxx_hal_conf.h

mysterywolf, 2021-01-10:
Hi, today, just now,  we released bluepill board bsp. You can re-create a project by using "Base on board".
[image]

M0n7y5, 2021-01-11:
Oh nice! I am gonna try it  ASAP! If thats gonna work i will close this issue.

M0n7y5, 2021-01-12:
Ok, so i am unable to get USB CDC workign with provided BSP. I checked configuration of clocks and rt-thread settings, but for some reason i get Unrecognized Device every time i plug it into PC. However i was able to get USB working using Chip based configuration instead (look above). 

So this is my configuration in STM32Cube:
[image]

[image]

M0n7y5, 2021-01-12:
So i am finally able to use USB CDC and finsh via UART. But only if i use project based on CHIP configuration. I had no luck with released BluePill BSP. 

[image]

M0n7y5, 2021-01-13:
@mysterywolf Can u test USB CDC on your side?

mysterywolf, 2021-01-13:
Hi, Today, @balanceTWK  has added the USB CDC function for your need. But please wait for a moment, I need some time to carry this function into RT-Studio version. And I will release a tutorial to help you how to create a bluepill project with USB CDC. And, please tell you about your next purpose, we could help you to develop because bluepill board is a very popular board.

mysterywolf, 2021-01-13:
And I will translate the vconsole to English for you.

mysterywolf, 2021-01-15:
vconsole English version https://github.com/enkiller/vconsole/pull/3

mysterywolf, 2021-01-18:
Hi, today, we release the bluepill rt-studio project that allows you can use USB CDC as a console device. I shot a video for you to teach you how to create a USB CDC project by using rt-studio.
[Uploading 20210118_183810.zip…]()

mysterywolf, 2021-01-18:
Hi, today, we release the bluepill rt-studio project that allows you can use USB CDC as a console device. I shot a video for you to teach you how to create a USB CDC project by using rt-studio.
[20210118_183810.zip](https://github.com/RT-Thread/rt-thread/files/5829692/20210118_183810.zip)

M0n7y5, 2021-01-18:
Nice! I am gonna try it asap.

M0n7y5, 2021-01-18:
Also it would be really good to upload that video on official RT-Thread Youtube channel :)

mysterywolf, 2021-01-18:
Yes, This video will upload on youtube and Twitter. Be advised, please pay attention to the ROM and RAM. I think there no extra space(ROM) to create a new thread if you add the function of USB-CDC virtual com.

M0n7y5, 2021-01-18:
@mysterywolf Oh i forgot to mention... I am using BluePill with 128KB Flash memory. So don't worry about that :)

mysterywolf, 2021-01-18:
`list_thread` and `free`. These two commands are very useful to reduce RAM.

closed 2021-02-05 by mysterywolf