RT-Thread/rt-thread #4239: USB MSC host stuck in STM32F746ng-disco

updesh31, 2021-01-15:
after some steps USB MSC HOST stuck in STM32F746
Out-put log:

cd /USB
msh /USB>ls
Directory /USB:
rt_usbh_storage_read10
pipe transform remain size,: 31
ok
pipe transform remain size,: 512
ok
pipe transform remain size,: 448
ok
pipe transform remain size,: 384
ok
pipe transform remain size,: 320
ok
pipe transform remain size,: 256
ok
pipe transform remain size,: 192
ok
pipe transform remain size,: 128
ok
pipe transform remain size,: 64
ok
pipe transform remain size,: 13
ok
System Volume Information<DIR>
setup.ini           7402
BACD99888           <DIR>
00000255            <DIR>
GWRDAWS1            <DIR>
COMFIG.ini          504
Accinfo1.csv        19052
DeducItem.csv       141
UserAcc.csv         12921
COMFIGSD.ini        508
rt_usbh_storage_read10
pipe transform remain size,: 31
ok
pipe transform remain size,: 512
stall
request size 512, transfer size 0

BernardXiong, 2021-01-17:
then?

updesh31, 2021-01-17:
Never exit from this function..............................
static int drv_pipe_xfer(upipe_t pipe, rt_uint8_t token, void *buffer, int nbytes, int timeouts)
{
    int timeout = timeouts;

    while (1)
    {
        if (!connect_status)
        {
            return -1;
        }
        rt_completion_init(&urb_completion);
        HAL_HCD_HC_SubmitRequest(&stm32_hhcd_fs,
                                 pipe->pipe_index,
                                 (pipe->ep.bEndpointAddress & 0x80) >> 7,
                                 pipe->ep.bmAttributes,
                                 token,
                                 buffer,
                                 nbytes,
                                 0);
        rt_completion_wait(&urb_completion, timeout);
        rt_thread_mdelay(50);
        if (HAL_HCD_HC_GetState(&stm32_hhcd_fs, pipe->pipe_index) == HC_NAK)
        {
            RT_DEBUG_LOG(RT_DEBUG_USB, ("nak\n"));
            if (pipe->ep.bmAttributes == USB_EP_ATTR_INT)
            {
                rt_thread_delay((pipe->ep.bInterval * RT_TICK_PER_SECOND / 1000) > 0 ? (pipe->ep.bInterval * RT_TICK_PER_SECOND / 1000) : 1);
            }
            HAL_HCD_HC_Halt(&stm32_hhcd_fs, pipe->pipe_index);
            HAL_HCD_HC_Init(&stm32_hhcd_fs,
                            pipe->pipe_index,
                            pipe->ep.bEndpointAddress,
                            pipe->inst->address,
                            USB_OTG_SPEED_FULL,
                            pipe->ep.bmAttributes,
                            pipe->ep.wMaxPacketSize);
            continue;
        }
        else if (HAL_HCD_HC_GetState(&stm32_hhcd_fs, pipe->pipe_index) == HC_STALL)
        {
            RT_DEBUG_LOG(RT_DEBUG_USB, ("stall\n"));
            pipe->status = UPIPE_STATUS_STALL;
            if (pipe->callback != RT_NULL)
            {
                pipe->callback(pipe);
            }
            return -1;
        }
        else if (HAL_HCD_HC_GetState(&stm32_hhcd_fs, pipe->pipe_index) == URB_ERROR)
        {
            RT_DEBUG_LOG(RT_DEBUG_USB, ("error\n"));
            pipe->status = UPIPE_STATUS_ERROR;
            if (pipe->callback != RT_NULL)
            {
                pipe->callback(pipe);
            }
            return -1;
        }
        else if(URB_DONE == HAL_HCD_HC_GetURBState(&stm32_hhcd_fs, pipe->pipe_index))
        {
            RT_DEBUG_LOG(RT_DEBUG_USB, ("ok\n"));
            pipe->status = UPIPE_STATUS_OK;
            if (pipe->callback != RT_NULL)
            {
                pipe->callback(pipe);
            }
            size_t size = HAL_HCD_HC_GetXferCount(&stm32_hhcd_fs, pipe->pipe_index);
            if (pipe->ep.bEndpointAddress & 0x80)
            {
                return size;
            }
            else if (pipe->ep.bEndpointAddress & 0x00)
            {
                return size;
            }
            return nbytes;
        }
        continue;
    }
}

BernardXiong, 2021-01-17:
run ls cmd, then not exit? How about your U disk?

updesh31, 2021-01-17:
first time Udisk initialized successfully but after executing any operation on disk it stops working.

updesh31, 2021-01-17:
I am always getting "request size 512, transfer size 0".

closed 2022-01-23 by mysterywolf

updesh31, 2022-01-23:
Now I am able to use both usb as host with hal library in rt-thread.