RT-Thread/rt-thread #4247: [UFFS][SPINAND] Without touching spare area?

wosayttn, 2021-01-18:
Hi,

We are working a SPI NAND flash(2048+64) driver based on RT_Device_Class_MTD and use DFS_UFFS as its file system.

(1) The SPI NAND device uses its HWECC to calculate parity code and write in spare area.  
     It implies that no available OOB area size for upper file system.  ( rt_mtd_nand_device.oob_free=0; )
     UFFS can support without spare-writing, like YAFFS's inband-tags function.
         
(2) We can burn an UFFS image using burner before accessing in RTT OS?
     -> if yes, how to make the UFFS image for 2048+64 SPI NAND?

BernardXiong, 2021-01-19:
Maybe you need to ask the author of uffs.

wosayttn, 2021-01-20:
Thanks for your comment.

closed 2021-01-20 by wosayttn