RT-Thread/rt-thread #4257: finsh和console以不同标志重复打开相同设备可能导致的潜在BUG

burytheflower, 2021-01-21:
## finsh和console以不同标志重复打开相同设备可能导致的潜在BUG
finsh和console(ulog基于console)默认使用同一个设备，二者会重复打开，问题如下：
```c
static rt_err_t rt_serial_open(struct rt_device *dev, rt_uint16_t oflag)
{
    // ... ignore some code

    /**
     * BUG: 重复打开串口设备将会清空之前的标志。
     * 
     * 例如首次打开是以RT_DEVICE_FLAG_DMA_TX标志，第二次是以轮询发送方式打开，
     * 则之前的RT_DEVICE_FLAG_DMA_TX标志将会被清除，而硬件上实际是配置成了DMA发送，这样会导致发送时卡死在轮询实现函数上。
     * 
     * finsh和console默认使用同一个设备，将会导致重复打开。
     * 目前finsh在console之后初始化，并只是其基础上增加了RT_DEVICE_FLAG_INT_RX标志，二者未发生冲突，所以目前不会出问题。
     */
    /* get open flags */
    dev->open_flag = oflag & 0xff;
    
    // ... ignore some code
}
```
之前在rt_hw_board_init()中添加了一段修改console设备为DMA发送的代码，然后就卡在了stm32_putc()中，该函数是finsh间接调用的。

BernardXiong, 2021-01-21:
是的，因为一个device节点是单一的，所以这部分没什么好的办法来区分。也许未来可以在open device时创建副本，不过这样在内存占用上会增加

burytheflower, 2021-01-21:
> 是的，因为一个device节点是单一的，所以这部分没什么好的办法来区分。也许未来可以在open device时创建副本，不过这样在内存占用上会增加

可能多个device节点也无法解决这个问题，因为实际硬件配置是唯一的，而多个device节点保存的打开标志可能与实际硬件配置不一致甚至不兼容，毕竟DMA_TX/INT_TX/DMA_RX/INT_RX这些硬件相关的参数也被引入到了打开标志中

i-jaffer, 2022-12-12:
与控制台无法输入相关的同类问题：修复低概率终端无法输入问题 #6464

closed 2024-03-29 by mysterywolf