RT-Thread/rt-thread #4262: [Question] Possible support for Raspberry Pi Pico?

M0n7y5, 2021-01-22:
Hi,
is it possible to add support for upcoming new cheap board from Raspberry Pi?

More info:
- https://www.raspberrypi.org/blog/raspberry-pi-silicon-pico-now-on-sale/
- https://www.raspberrypi.org/products/raspberry-pi-pico/
- https://github.com/raspberrypi/pico-sdk

mysterywolf, 2021-01-22:
Yes, we have noticed that news, and we will plan to work on it.

mysterywolf, 2021-01-22:
I have a question: where can you buy bluepill board and Raspberry Pi Pico in your country? Can you give me a shopping web site link?

M0n7y5, 2021-01-22:
Oh yeah no problem ...
For specifically RPi stuff, there is this store:
- https://rpishop.cz/pico/3352-raspberry-pi-pico-0617588405587.html

For other stuff (where you can also buy RPi stuff) there is 
- https://www.laskarduino.cz/bluepill-arm-stm32-stm32f103c8-vyvojova-deska/
- https://pajenicko.cz/vyvojova-deska-stm32-bluepill-s-stm32f103c8t6
- http://www.automatizace.kvalitne.cz/vyvojove-desky/105-stm32f103c8t6-arm-stm32-vyvojova-deska-pouzitelna-s-arduino-pcb-typ-b.html
 
Also, good amount of people here are also buying from stores like Amazon, Aliexpress or Banggood ...

liu2guang, 2021-01-28:
https://github.com/Guozhanxin/rt-thread/tree/raspberry-pico/bsp/raspberry-pico

[image]

yoyojacky, 2021-01-28:
[image]
RT-Thread single kernel running is good.

BernardXiong, 2021-01-28:
Merged with #4275 !

Who can provide the mpy with RT-Thread? Welcome!

mysterywolf, 2021-01-29:
https://twitter.com/rt_thread/status/1354730387155480578

mysterywolf, 2021-02-04:
https://rt-thread.medium.com/getting-started-with-raspberry-pi-pico-in-rt-thread-studio-ide-5bb08c01363f

mysterywolf, 2021-02-06:
https://club.rt-thread.org/ask/article/2539.html

closed 2021-06-07 by mysterywolf