RT-Thread/rt-thread #4278: 汇编文件.S无法被Keil接受！！！

studyHooligen, 2021-01-29:
我使用的是Keil5.28版本，在使用新的“default compiler Version 6”时发现，编译器会将目前RTT的内核汇编文件“context_rvds.S"识别为C/C++文件。我尝试将后缀改为”.s“就可以正常被编译器当作汇编文件了
希望RTT团队尽早解决该问题

yangjie11, 2021-01-30:
应该是识别文件类型错误导致的，而不是大小写影响的类型识别错误。

我这边复现不出来，不论大小写都能正确识别汇编文件。请问你那边还能复现吗，可以联系我

studyHooligen, 2021-01-30:
[image]
我使用的是Version 6编译器，编译报错如下图：
[image]
显然是把”.S“后缀的文件当作了C文件编译

yangjie11, 2021-01-30:
我验证的是直接使用keil添加某个汇编文件，这样识别是没问题的。
看上图，你是用cube生成的keil代码，我先去复现下再看下是什么问题

yangjie11, 2021-01-30:
你好，检查了下，我们在打包的时候，确认该汇编文件是汇编文件类型，但最后在cube生成的keil工程中文件类型被改变为c文件类型。像是cube导出keil工程后，帮keil决定的文件类型。这个需要和st cube沟通一下。

使用AC6，你可以将该s文件类型手动修改为汇编，然后编译。感谢反馈。
修改：右击改文件，修改文件类型

[image]

closed 2021-02-06 by mysterywolf