RT-Thread/rt-thread #4279: 仓库中还有很多文件依然保留老版本RTT使用的GNU General Public协议，会对用户使用造成误解

mysterywolf, 2021-01-30:
```c
/*
 * File      : main.c
 * This file is part of RT-Thread RTOS
 * COPYRIGHT (C) 2006 - 2018, RT-Thread Development Team
 *
 *  This program is free software; you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation; either version 2 of the License, or
 *  (at your option) any later version.
 *
 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License along
 *  with this program; if not, write to the Free Software Foundation, Inc.,
 *  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301 USA.
 *
 * Change Logs:
 * Date           Author       Notes
 */

```

BernardXiong, 2021-01-31:
这部分需要梳理的了，而不是一刀切

mysterywolf, 2021-01-31:
没有一刀切的，一部分第三方的没有动，可以再筛一遍

BernardXiong, 2021-02-01:
可以提供个列表出来，另外，在BSP目录下还需要一个许可协议声明，厂商外设库是需要按照各自的许可协议方式的。

closed 2021-02-22 by mysterywolf