RT-Thread/rt-thread #4281: getting following issue after running scons --menuconfig. and where  do i install env tool?

UchihaSR, 2021-01-30:
~/iot/rt-thread/bsp/qemu-vexpress-a9 $ scons --menuconfig
scons: Reading SConscript files ...
AttributeError: 'NoneType' object has no attribute 'groups':
  File "/home/itachi/iot/rt-thread/bsp/qemu-vexpress-a9/SConstruct", line 29:
    objs = PrepareBuilding(env, RTT_ROOT)
  File "/home/itachi/iot/rt-thread/bsp/qemu-vexpress-a9/../../tools/building.py", line 338:
    PreProcessor.process_contents(contents)
  File "/usr/lib/scons/SCons/cpp.py", line 307:
    self.tuples = self.tupleize(contents)
  File "/usr/lib/scons/SCons/cpp.py", line 287:
    return  [(m[0],) + Table[m[0]].match(m[1]).groups() for m in cpp_tuples]
  File "/usr/lib/scons/SCons/cpp.py", line 287:
    return  [(m[0],) + Table[m[0]].match(m[1]).groups() for m in cpp_tuples]

mysterywolf, 2021-01-30:
You can install Env tool at here : https://www.rt-thread.io/download.html?download=Env
and this is the reference: https://github.com/RT-Thread/rtthread-manual-doc/blob/master/env/env.md

mysterywolf, 2021-01-31:
Linux version: https://supperthomas-wiki.readthedocs.io/en/latest/04_dozingfiretruck/06/Linux.html

closed 2021-02-26 by mysterywolf