RT-Thread/rt-thread #4309: rt_timer_list_next_timeout() may return error?

kevin-is-kevin, 2021-02-05:
If a timer has a timeout tick 0xFFFFFFFF(RT_TICK_MAX),  the timer thread will suspend itself.  And if the rt_tick now is 0xFFFFFFFF(RT_TICK_MAX), timer callback should be called immediately. 

```C

  /* the fist timer always in the last row */
  static rt_tick_t rt_timer_list_next_timeout(rt_list_t timer_list[])
  {
      struct rt_timer *timer;
      register rt_base_t level;
      rt_tick_t timeout_tick = RT_TICK_MAX;
  
      /* disable interrupt */
      level = rt_hw_interrupt_disable();
  
      if (!rt_list_isempty(&timer_list[RT_TIMER_SKIP_LIST_LEVEL - 1]))
      {
          timer = rt_list_entry(timer_list[RT_TIMER_SKIP_LIST_LEVEL - 1].next,
                                struct rt_timer, row[RT_TIMER_SKIP_LIST_LEVEL - 1]);
          timeout_tick = timer->timeout_tick;
      }
  
      /* enable interrupt */
      rt_hw_interrupt_enable(level);
  
      return timeout_tick;
  }

```

```C

        /* get the next timeout tick */
        next_timeout = rt_timer_list_next_timeout(rt_soft_timer_list);
        if (next_timeout == RT_TICK_MAX)
        {
            /* no software timer exist, suspend self. */
            rt_thread_suspend(rt_thread_self());
            rt_schedule();
        }

```

Is that OK?

I suppose that the API should be "rt_err_t rt_timer_list_next_timeout(rt_list_t timer_list[], rt_tick_t *timeout_tick) " instead,

Thanks.

BernardXiong, 2021-02-07:
The max timeout tick should be UINT32_MAX / 2.

kevin-is-kevin, 2021-02-07:
@BernardXiong    I have made a test on this, here is the code:

```C
/**

rt_timer_t test;

void timeout(void * p)
{

}

int main(void)
{
    /* create a timer and make the timeout tick RT_TICK_MAX */
    test = rt_timer_create("test", timeout, NULL, 10,
            RT_TIMER_FLAG_PERIODIC | RT_TIMER_FLAG_SOFT_TIMER);
    rt_tick_set(RT_TICK_MAX - 10);
    rt_timer_start(test);

    while (1)
    {
        rt_thread_delay(100);
    }
}

```

RT_ASSERT was added int function rt_timer_list_next_timeout:

```C

/* the fist timer always in the last row */
static rt_tick_t rt_timer_list_next_timeout(rt_list_t timer_list[])
{
    struct rt_timer *timer;
    register rt_base_t level;
    rt_tick_t timeout_tick = RT_TICK_MAX;

    /* disable interrupt */
    level = rt_hw_interrupt_disable();

    if (!rt_list_isempty(&timer_list[RT_TIMER_SKIP_LIST_LEVEL - 1]))
    {
        timer = rt_list_entry(timer_list[RT_TIMER_SKIP_LIST_LEVEL - 1].next,
                              struct rt_timer, row[RT_TIMER_SKIP_LIST_LEVEL - 1]);
        timeout_tick = timer->timeout_tick;

        /* we will check the timeout value here */
        RT_ASSERT(timeout_tick != RT_TICK_MAX);
    }

    /* enable interrupt */
    rt_hw_interrupt_enable(level);

    return timeout_tick;
}

```

The timeout tick was set like this :

```C

    /*
     * get timeout tick,
     * the max timeout tick shall not great than RT_TICK_MAX/2
     */
    RT_ASSERT(timer->init_tick < RT_TICK_MAX / 2);
    timer->timeout_tick = rt_tick_get() + timer->init_tick;

```

The "timer->timeout_tick" will always have chance to reach RT_TICK_MAX, so the API will return a wrong tick

sexenteen, 2026-11-01:
This looks resolved on current master (RT-Thread 5.3.1): the API was changed to the shape
suggested here, and the boundary case now behaves correctly. Three pieces of evidence below.

### 1) The API shape is what this issue proposed

`src/timer.c` now returns the status instead of overloading `RT_TICK_MAX` as a sentinel:

```c
static rt_err_t _timer_list_next_timeout(rt_list_t timer_list[], rt_tick_t *timeout_tick)
{
    ...
    *timeout_tick = timer->timeout_tick;
    return RT_EOK;
    ...
    return -RT_ERROR;   /* empty list */
}
```

and the caller uses the return value rather than comparing against `RT_TICK_MAX`:

```c
ret = _timer_list_next_timeout(_soft_timer_list, &next_timeout);
if ((ret == RT_EOK) && (next_timeout <= rt_tick_get()))
    rt_sem_release(&_soft_timer_sem);
```

The soft-timer thread is also no longer suspended on `next_timeout == RT_TICK_MAX`:
`_timer_thread_entry()` waits on `_soft_timer_sem` with `RT_WAITING_FOREVER`.

### 2) The scenario from this issue was re-run and passes

`bsp/qemu-vexpress-a9`, `RT_TICK_PER_SECOND=100`, `RT_USING_TIMER_SOFT=y`. Tick set near the
maximum and a 10-tick software timer, so that `timeout_tick` lands exactly on `RT_TICK_MAX`
(the value that used to mean "no timer"):

```
[B2] tick after  set = 4294967285 (expected 4294967285)
[B2] started: init_tick=10 timeout_tick=4294967295 (RT_TICK_MAX=4294967295)
[B2] callback fired: fired=1 tick=4294967295
[B2] RESULT: PASS - timer fired across the tick wrap
```

So a real timer at `RT_TICK_MAX` is no longer confused with an empty timer list.

### 3) The limit suggested in the discussion is enforced and documented

`rt_timer_init()`, `rt_timer_create()` and `rt_timer_control(RT_TIMER_CTRL_SET_TIME)` all
assert `time < RT_TICK_MAX / 2`, and the doxygen comment states
"The max timeout tick should be no more than (RT_TICK_MAX/2 - 1)". Passing `RT_TICK_MAX` as a
duration trips the assert:

```
(time < RT_TICK_MAX / 2) assertion failed at function:rt_timer_create, line number:355
```

### Minor observation (not a defect)

`rt_timer_check()` compares `next_timeout <= rt_tick_get()` without wrap handling, so around
the tick wrap point `_soft_timer_sem` can be released early. That only costs an extra
`_timer_check()` pass, and `_timer_check()` itself compares wrap-safely
(`(current_tick - t->timeout_tick) < RT_TICK_MAX / 2`), so no timer fires early or late.
Using `(rt_tick_t)(next_timeout - rt_tick_get()) < RT_TICK_MAX / 2` would be cosmetic only.

Since the reported problem no longer reproduces, I think this issue can be closed. Please let
me know if you would rather keep it open to track anything else.