RT-Thread/rt-thread #4323: [BUG]在旧版本代码(无SAL版本）中应用lwip2.0.2， 当client连接个数大于配置个数时，net_sockets.c报异常导致程序异常终止退出

ChouJoe, 2021-02-08:
### 该问题是怎么引起的？
我在rtt中基于lwip2.0.2版本应用modbustcp应用程序，根据网上lwip移植配置方法，我在rtt中成功实现modbustcp 作为slave运行， 但是当我连接的client数量变多时，我发现程序报错并异常终止shutdown。 根据我的理解，能连接的client的最大数量和rtconfig.h里的配置有关，我在rtconfig.h中的配置如下：
[image]
我可以理解当client数量大于配置数量时，程序会报错， 但是不应该退出终止程序。以下是报错的调用堆栈信息，最后报错出现在rt-thread\components\dfs\filesystems\net\net_sockets.c的event_callback函数中的rt_wqueue_wakeup(&sock->wait_head, (void*)event);我不知道该如何解决这个问题，请问有人可以帮助解决吗？
### 重现步骤
出错的函数：
```
static void event_callback(struct netconn *conn, enum netconn_evt evt, u16_t len)
{
    int s;
    struct lwip_sock *sock;
    uint32_t event = 0;
    SYS_ARCH_DECL_PROTECT(lev);

    LWIP_UNUSED_ARG(len);

    /* Get socket */
    if (conn)
    {
        s = conn->socket;
        if (s < 0)
        {
            /* Data comes in right away after an accept, even though
             * the server task might not have created a new socket yet.
             * Just count down (or up) if that's the case and we
             * will use the data later. Note that only receive events
             * can happen before the new socket is set up. */
            SYS_ARCH_PROTECT(lev);
            if (conn->socket < 0)
            {
                if (evt == NETCONN_EVT_RCVPLUS)
                {
                    conn->socket--;
                }
                SYS_ARCH_UNPROTECT(lev);
                return;
            }
            s = conn->socket;
            SYS_ARCH_UNPROTECT(lev);
        }

        sock = lwip_tryget_socket(s);
        if (!sock)
        {
            return;
        }
    }
    else
    {
        return;
    }

    SYS_ARCH_PROTECT(lev);
    /* Set event as required */
    switch (evt)
    {
    case NETCONN_EVT_RCVPLUS:
        sock->rcvevent++;
        break;
    case NETCONN_EVT_RCVMINUS:
        sock->rcvevent--;
        break;
    case NETCONN_EVT_SENDPLUS:
        sock->sendevent = 1;
        break;
    case NETCONN_EVT_SENDMINUS:
        sock->sendevent = 0;
        break;
    case NETCONN_EVT_ERROR:
        sock->errevent = 1;
        break;
    default:
        LWIP_ASSERT("unknown event", 0);
        break;
    }

    if (sock->lastdata || sock->rcvevent > 0) event |= POLLIN;
    if (sock->sendevent) event |= POLLOUT;
    if (sock->errevent)  event |= POLLERR;

    SYS_ARCH_UNPROTECT(lev);

    if (event)
    {
    	if(sock == NULL){
    		return -1;
    	}
        rt_wqueue_wakeup(&sock->wait_head, (void*)event);
    }
}

```
串口打印的报错信息

```

init_dev_diag enter
watch dog rd sts: 0
watch dog wt sts: 0
init_dev_diag exit
start_IpcInit enter
start_IpcInit exit
start_RpcInit enter
start_RpcInit exit
msh />link change: DOWN
vbus initialization
VBUS for CommIPC initialization
init_cpuload enter
init_cpuload exit
MainInit enter
MainInit exit
UserApplicationInit enter
UserApplicationInit exit
link change: UP
CLIENT_SOCK_FD:5
CLIENT_SOCK_FD:6
CLIENT_SOCK_FD:7
CLIENT_SOCK_FD:8
CLIENT_SOCK_FD:9
CLIENT_SOCK_FD:10
DFS fd new is failed! Could not found an empty fd entry.CLIENT_SOCK_FD:-1
data abort:Execption:
r00:0x60000013 r01:0x600000d3 r02:0x00000000 r03:0xfffffffc
r04:0x00000000 r05:0x00000000 r06:0x00000000 r07:0x00000000
r08:0x00000000 r09:0x00000000 r10:0x00000000
fp :0x1c3ed574 ip :0x1c3e8fa6
sp :0x1c3ed550 lr :0x1c0e4d6c pc :0x1c0e4d98
cpsr:0x80000093
  DFSR: 00001008
  IFSR: 00000000
  ADFSR: 00000000
  AIFSR: 00000000
  DFAR: 00000008
  IFAR: 00000000
[720]: tcpip -> tidle
[721]: tidle -> erx
[722]: erx -> tcpip
[723]: tcpip -> erx
[724]: erx -> etx
[725]: etx -> tcpip
[726]: tcpip -> etx
[727]: etx -> tidle
[728]: tidle -> erx
[729]: erx -> tcpip
[730]: tcpip -> mbtcp_
[731]: mbtcp_ -> erx
[732]: erx -> tidle
[733]: tidle -> erx
[734]: erx -> tcpip
watch dog wt sts: 8

Set_Exception_Info: len:972, type:4
       pri  status      sp     stack size max used left tick  error
------ ---  ------- ---------- ----------  ------  ---------- ---
tBkgTs  15  suspend 0x000001a8 0x00002000    05%   0x00000014 000
tRpcIn  29  suspend 0x00000218 0x00002004    08%   0x00000012 000
tIpcIn  29  suspend 0x0000021c 0x00002000    08%   0x00000012 000
rshell  25  suspend 0x000001fc 0x00000800    24%   0x00000014 000
tshell  20  suspend 0x000001d0 0x00002000    08%   0x0000000a 000
mbtcp_  12  suspend 0x00000414 0x00002800    20%   0x00000007 000
vbusin   9  suspend 0x000001b8 0x00000800    21%   0x00000014 000
vbusou   8  suspend 0x000001c8 0x00000800    22%   0x00000014 000
phyDct  29  suspend 0x000001b4 0x00000400    64%   0x00000011 000
tcpip   12  ready   0x000001f0 0x00002800    23%   0x00000014 000
etx     14  suspend 0x000001b4 0x00002800    04%   0x00000010 000
erx     14  ready   0x0000020c 0x00002800    08%   0x0000000b 000
tidle   31  ready   0x0000017c 0x00001000    09%   0x0000001a 000
Current thread: tcpip
----------------------------------------------------------
  stack  | 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
----------------------------------------------------------
1c3ed550 | 03 00 00 00 cc c4 3a 1c 00 00 00 00 fc ff ff ff
1c3ed560 | 13 00 00 60 00 00 00 00 3c 03 10 1c 3c 03 10 1c
1c3ed570 | 94 d5 3e 1c d4 04 0e 1c 0c 00 3e 00 34 5d 3b 1c
1c3ed580 | 94 d5 3e 1c b4 c4 3a 1c 03 00 00 00 0e 00 00 00
1c3ed590 | b4 d5 3e 1c 10 05 10 1c 4e 94 3e 00 0c 94 3e 1c
1c3ed5a0 | 98 6a 3b 1c 34 5d 3b 1c 34 5d 3b 1c 00 00 0c 00
1c3ed5b0 | f4 d5 3e 1c 14 24 11 1c d8 48 3b 1c c0 a8 01 26
1c3ed5c0 | 98 d6 3e 1c 0c 94 3e 1c 00 00 62 00 f4 d5 3e 14
1c3ed5d0 | 00 00 ec 00 00 00 00 00 00 00 00 00 00 00 14 00
1c3ed5e0 | 00 00 00 00 98 6a 3b 1c 93 00 ec ff 00 00 00 00
1c3ed5f0 | 24 d6 3e 1c d4 18 12 1c 98 d6 3e 1c 0c 94 3e 1c
1c3ed600 | 2c 00 f2 ff 0c 94 3e 1c 2c 94 3e 1c 34 00 14 00
1c3ed610 | 3a 94 3e 1c 01 00 00 00 01 00 00 00 98 d6 3e 1c
1c3ed620 | 44 d6 3e 1c fc 9b 11 1c 98 d6 3e 1c 0c 94 3e 1c
1c3ed630 | 54 d6 3e 1c c0 48 08 00 2c 94 3e 1c 13 00 0e 00
1c3ed640 | 5c d6 3e 1c a0 90 10 1c 00 00 00 00 00 00 00 00
1c3ed650 | 00 00 00 00 24 e5 3b 1c 00 00 00 00 10 1b 0d 1c
----------------------------------------------------------
backtrace:
please use: addr2line -e rtthread.elf -a -f 1c0e4d90 1c0e04d4 1c100510 1c112414 1c1218d4 1c119bfc 1c1090a0
----------------------------------------------------------
recent schedule recoder:
[720]: tcpip -> tidle
[721]: tidle -> erx
[722]: erx -> tcpip
[723]: tcpip -> erx
[724]: erx -> etx
[725]: etx -> tcpip
[726]: tcpip -> etx
[727]: etx -> tidle
[728]: tidle -> erx
[729]: erx -> tcpip
[730]: tcpip -> mbtcp_
[731]: mbtcp_ -> erx
[732]: erx -> tidle
[733]: tidle -> erx
[734]: erx -> tcpip
----------------------------------------------------------
shutdown...

```

### 报错信息

[image]

Guozhanxin, 2021-02-22:
@xiangxistu please take a look, thanks!

xiangxistu, 2021-02-22:
CPSR: 80000093
I think you should notice the CPSR register.
From the cpsr info, system abort because manage fault.

Your can modify 'cpuport.c' according the laster of RT-Thread code to get clear hardfault information.
Throught Cmbracktarce to find wrong code that cause hard frault.

closed 2022-01-23 by mysterywolf