RT-Thread/rt-thread #4355: [ifconfig] Wrong networking interface information issue.

wosayttn, 2021-02-19:
Hi,

    I use ifconfig command to list dual-ethernet interfaces as below, but dump information is incorrect.

msh />ifconfig
network interface device: e0 (Default)
MTU: 1500
MAC: 82 06 21 53 94 53
FLAGS: UP LINK_UP INTERNET_DOWN DHCP_ENABLE ETHARP BROADCAST IGMP
ip address: 192.168.1.199
gw address: 192.168.1.1
net mask  : 255.255.255.0
dns server #0: 192.168.31.1
dns server #1: 10.1.8.12

network interface device: e1
MTU: 1500
MAC: 82 06 21 53 94 54
FLAGS: DOWN LINK_DOWN INTERNET_DOWN DHCP_ENABLE
ip address: 0.0.0.0
gw address: 0.0.0.0
net mask  : 0.0.0.0
dns server #0: 0.0.0.0
dns server #1: 0.0.0.0
msh />
msh />ping 192.168.31.1
dhcp_arp_reply()
60 bytes from 192.168.31.1 icmp_seq=0 ttl=64 time=1 ms
60 bytes from 192.168.31.1 icmp_seq=1 ttl=64 time=0 ms
60 bytes from 192.168.31.1 icmp_seq=2 ttl=64 time=0 ms
60 bytes from 192.168.31.1 icmp_seq=3 ttl=64 time=0 ms
msh />
msh />ping 192.168.1.1
60 bytes from 192.168.1.1 icmp_seq=0 ttl=64 time=1 ms
60 bytes from 192.168.1.1 icmp_seq=1 ttl=64 time=0 ms
60 bytes from 192.168.1.1 icmp_seq=2 ttl=64 time=0 ms
60 bytes from 192.168.1.1 icmp_seq=3 ttl=64 time=0 ms
msh />

Guozhanxin, 2021-02-22:
@xiangxistu please take a look, thanks!

xiangxistu, 2021-02-22:
According to your log, your network device both are ETH device.
For ETH device, The network information will be provided by lwip protocol stack.

I think you can direct get information of lwip netif[0] and netif[1].
the possible reason:
1. The log is too old, you should refresh it. Maybe when you send “ifconfig” command，the "e1" haven't been worked at that time.
2. netif[1] haven't work normally,  your code for dual-ethernet is error.

wosayttn, 2021-02-23:
Here, bug->[(rt_strncmp(netdev->name, name, RT_NAME_MAX) == 0)](https://github.com/RT-Thread/rt-thread/blob/master/components/net/netdev/src/netdev.c#L263)

Revert to previous version -> (rt_strncmp(netdev->name, name, rt_strlen(netdev->name))
it's fine on our platform.

msh />ifconfig
network interface device: e0 (Default)
MTU: 1500
MAC: 82 06 21 53 94 53
FLAGS: UP LINK_UP INTERNET_DOWN DHCP_ENABLE ETHARP BROADCAST IGMP
ip address: 192.168.1.199
gw address: 192.168.1.1
net mask  : 255.255.255.0
dns server #0: 192.168.31.1
dns server #1: 10.1.8.12

network interface device: e1
MTU: 1500
MAC: 82 06 21 53 94 54
FLAGS: UP LINK_UP INTERNET_DOWN DHCP_ENABLE ETHARP BROADCAST IGMP
ip address: 192.168.31.135
gw address: 192.168.31.1
net mask  : 255.255.255.0
dns server #0: 192.168.31.1
dns server #1: 10.1.8.12

xiangxistu, 2021-02-24:
Thanks for your help, i think this reason is the length of name is different between lwip and rt-thread.
the length for lwip netif's name is 2, but rt-thread is 8.

--
Can you help me to verify my guess?

```c
rt_strncmp(netdev->name, **name**, rt_strlen(netdev->name)
```

Please show the origin data of 'name' parameter. 
I think for the network interface device'e0', the origin data will be ``` e00···```. For the 'e1', the origin data will be ```e11···```.

xiangxistu, 2021-02-24:
[image]

OK, Thanks for your help. We have find the final reason. 
We will fix the bug, thanks.

closed 2021-02-26 by mysterywolf