RT-Thread/rt-thread #4400: why save year month day information in bakeup register with stm32 rtc driver ?

iysheng, 2021-02-28:
The RTC peripheral in stm32 chip use one 32bit counter register to save time. the max value is 4294967295 secs, max value is 187 years more. When I read the rtc driver code with stm32, I wonder why saving year month date value in backup registers in function **set_rtc_time_stamp** with file bsp/stm32/libraries/HAL_Drivers/drv_rtc.c.Just for keeping up with the official HAL libs of stm32?
```
#ifdef SOC_SERIES_STM32F1
    /* F1 series does't save year/month/date datas. so keep those datas to bkp reg */
    HAL_RTCEx_BKUPWrite(&RTC_Handler, RTC_BKP_DR2, RTC_DateStruct.Year);
    HAL_RTCEx_BKUPWrite(&RTC_Handler, RTC_BKP_DR3, RTC_DateStruct.Month);
    HAL_RTCEx_BKUPWrite(&RTC_Handler, RTC_BKP_DR4, RTC_DateStruct.Date);
    HAL_RTCEx_BKUPWrite(&RTC_Handler, RTC_BKP_DR5, RTC_DateStruct.WeekDay);
#endif
```

mysterywolf, 2021-03-15:
STM32F1出品时间较早，RTC实现方式和后续系列有较大不同，仅仅提供了一个计数器。后续是直接把年份什么的写在寄存器里，直接读就可以。因此F1没有年月日周的寄存器，为了和后续系列保持一致。

iysheng, 2021-03-15:
> STM32F1出品时间较早，RTC实现方式和后续系列有较大不同，仅仅提供了一个计数器。后续是直接把年份什么的写在寄存器里，直接读就可以。因此F1没有年月日周的寄存器，为了和后续系列保持一致。

噢，👌

closed 2021-03-15 by iysheng