RT-Thread/rt-thread #4415: www.rt-thread.org/document 有关看门狗部分测试代码出现了 idle 线程栈溢出

iysheng, 2021-03-03:
[看门狗测试代码](https://www.rt-thread.org/document/site/programming-manual/device/watchdog/watchdog/) 发现会出现 idle 线程栈溢出：
```msh />wdt_sample iwdt
01-01 00:00:-1 I/drv.wdt: threshold section [1, 26]
msh />feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!
 feed the dog!thread:tidle0 stack overflow

 \ | /
- RT -     Thread Operating System
 / | \     4.0.3 build Mar  3 2021
 2006 - 2021 Copyright by rt-thread team
msh />

```
idle 线程的 hook 函数为：
``` C
static void idle_hook(void)
{
    /* 在空闲线程的回调函数里喂狗 */
    rt_device_control(wdg_dev, RT_DEVICE_CTRL_WDT_KEEPALIVE, NULL);
    rt_kprintf("feed the dog!\n ");
}
```
如果删除 ** rt_kprintf("feed the dog!\n ");**，则不会出现栈溢出，idle 线程的栈初始在 **43%**，执行测试看门狗后稳定在 **75%** 附近，具体对比为：
```bash
 \ | /
- RT -     Thread Operating System
 / | \     4.0.3 build Mar  3 2021
 2006 - 2021 Copyright by rt-thread team
msh />
msh />
msh />
msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000009 000
tidle0    31  ready   0x00000070 0x00000100    43%   0x0000001e 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />idw
msh />idw  w  wd
wdt_sample
msh />wdt_sample iwdt
01-01 00:00:-1 I/drv.wdt: threshold section [1, 26]
msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000003 000
tidle0    31  ready   0x00000098 0x00000100    65%   0x00000002 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000008 000
tidle0    31  ready   0x00000070 0x00000100    75%   0x0000000a 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000004 000
tidle0    31  ready   0x00000058 0x00000100    75%   0x0000000c 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000001 000
tidle0    31  ready   0x00000060 0x00000100    75%   0x0000001b 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000006 000
tidle0    31  ready   0x00000058 0x00000100    75%   0x00000004 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />msh />ps
thread   pri  status      sp     stack size max used left tick  error
-------- ---  ------- ---------- ----------  ------  ---------- ---
tshell    20  running 0x00000180 0x00001000    15%   0x00000003 000
tidle0    31  ready   0x00000098 0x00000100    75%   0x00000009 000
timer      4  suspend 0x0000007c 0x00000200    24%   0x00000009 000
main      10  suspend 0x000000b4 0x00000800    30%   0x00000014 000
msh />
```

BernardXiong, 2021-03-03:
增大idle任务的栈

iysheng, 2021-03-04:
> 增大idle任务的栈

经过测试增大 idle 任务到栈从 256 到 1024 不会导致栈溢出。我有一个疑问为什么在 256 的时候，rt_kpritf 多次就会导致溢出呢？是内存释放的速度没有线程调度的速度快吗？:thinking:

mysterywolf, 2021-03-04:
@iysheng 因为rt_kprintf内部的局部变量有点多 就直接把栈压炸了，rt_kprintf已经是精简过的了。正常保持256没有问题，记得如果在idle里边打印东西的时候临时加大栈空间即可

closed 2021-03-09 by mysterywolf