RT-Thread/rt-thread #4467: drv_uart 串口驱动BUG，DMA发送直接卡死

Samwuzhitao, 2021-03-16:
**问题描述**：

在xxx的项目中，使用的双串口的硬件，MCU值是在中间做一个透传，将uart1接收的数据发送给uart3,将uart3发送出来的数据发回给uart1,在完成硬件驱动调试之后发现：**设备有概率会卡死的UART3中断中无法退出，上电之后直接发送指令个UART1，基本必现直接卡死**

```cpp
uart1 = rt_device_find("uart1");
uart3 = rt_device_find("uart3");

/* step3：控制串口设备。通过控制接口传入命令控制字，与控制参数 */
ret = rt_device_control(uart1, RT_DEVICE_CTRL_CONFIG, &config);
rt_kprintf("INIT: uart1 set  ret:%d\r\n", ret);
rt_thread_mdelay(10);
ret = rt_device_control(uart3, RT_DEVICE_CTRL_CONFIG, &config);
rt_kprintf("INIT: uart3 set  ret:%d\r\n", ret);
rt_thread_mdelay(10);

ret = rt_device_open(uart1, RT_DEVICE_FLAG_DMA_TX | RT_DEVICE_FLAG_DMA_RX);
rt_kprintf("INIT: uart1 open ret:%d\r\n", ret);
rt_thread_mdelay(10);
ret = rt_device_open(uart3, RT_DEVICE_FLAG_DMA_TX | RT_DEVICE_FLAG_DMA_RX);
rt_kprintf("INIT: uart3 open ret:%d\r\n", ret);
rt_thread_mdelay(10);

/* 创建 serial 线程 */
cmd_thd = rt_thread_create("cmd_thd", send_cmd, RT_NULL, 1024, 15, 3);
if (cmd_thd != RT_NULL)
{
  rt_thread_startup(cmd_thd);
  rt_kprintf("cmd_thd: rt_thread_startup!\r\n");
  rt_thread_mdelay(10);
}

ack_thd = rt_thread_create("ack_thd", send_ack, RT_NULL, 1024, 15, 3);
if (ack_thd != RT_NULL)
{
  rt_thread_startup(ack_thd);
  rt_kprintf("ack_thd: rt_thread_startup!\r\n");
  rt_thread_mdelay(10);

}

/* 启动wifi */
rt_pin_write(WIFI_PWR_PIN, PIN_LOW);

while (count++)
{
  rt_kprintf("[ %d ] cnt [c:%d a:%d]\r\n", count, cmd_cnt, ack_cnt);
  rt_thread_mdelay(500);
  rt_pin_write(APP_LED0_PIN, PIN_HIGH);
  rt_thread_mdelay(500);
  rt_pin_write(APP_LED0_PIN, PIN_LOW);
}
```

**问题解决**：

对比正常和不正常的核心寄存器：

```c
uint32_t isrflags   = READ_REG(huart->Instance->SR);   0xC2     0xC0
uint32_t cr1its     = READ_REG(huart->Instance->CR1);  0x215C   0x215C
uint32_t cr3its     = READ_REG(huart->Instance->CR3);  0x40     0x40 

// 异常确实是在 USART_SR_FE : Framing Error

// 但是此处的错误处理：USART_CR3_EIE 需要打开在打开错误中断的情况下才处理
/* UART frame error interrupt occurred -----------------------------------*/
if (((isrflags & USART_SR_FE) != RESET) && ((cr3its & USART_CR3_EIE) != RESET))
{
  huart->ErrorCode |= HAL_UART_ERROR_FE;
}
```

于是修改代码：

```c
if (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_TC) != RESET)
{
  if ((serial->parent.open_flag & RT_DEVICE_FLAG_DMA_TX) != 0)
  {
    /* 在处理完成数据之前，清除这个数据帧错误 */
    __HAL_UART_CLEAR_FEFLAG(&uart->handle);
    HAL_UART_IRQHandler(&(uart->handle));
  }
  else
  {
    UART_INSTANCE_CLEAR_FUNCTION(&(uart->handle), UART_FLAG_TC);
  }
}
```

**总结**：

实际在测试的时候发现，原本不开启wifi模块，这个这个映射的串口发送也不会有问题，这个镇错误就是开始上电那会输出的一些乱码导致的,这个乱码是wifi模块启动时候的信息，波特率不是115200

```c
\ | /
- RT -     Thread Operating System
 / | \     4.0.2 build Mar 16 2021
 2006 - 2019 Copyright by rt-thread team
INIT: uart1 set  ret:0
INIT: uart3 set  ret:0
INIT: uart1 open ret:0
INIT: uart3 open ret:0
cmd_thd: rt_thread_startup!
ack_thd: rt_thread_startup!
[ 100001 ] cnt [c:20 a:9]
rl l湊r宐鈔?€l?p寍巶?靝挒帓滌lb宲~騨?lnn?bbp鞄bl`rlp騨鄠?値€lbn忖n鞄?b寍~騨?l寧 l`?nnl`nr彃瀝l渞lp騨鄠?s寽挏?巰bn忖n??b寍~騨?l`?
nnl`or帓瀝鞄rlp騨鄠?r寽挏?帉bn汊n??庎b寍騨?l`nnlnnl?rlnr彃瀝l巖lr挒rl巖l
[ 100002 ] cnt [c:1018 a:1013]
[ 100003 ] cnt [c:2017 a:2012]
[ 100004 ] cnt [c:3016 a:3011]
[ 100005 ] cnt [c:4015 a:4010]
```

实际上如果以中断方式传输数据也是没有问题的，单如果是以DMA方式传输数据，此处就会大概率的进入异常，导致中断无法清除，而一直在中断中，导致数据异常

BernardXiong, 2021-04-05:
是不是配置的问题？ `rt_device_control(uartx, RT_DEVICE_CTRL_CONFIG, &config);`

majianjia, 2021-05-29:
@BernardXiong @mysterywolf 
现阶段STM32的UART DMA 接收驱动应该是有问题。
我这里遇到DMA接收不完整的问题。
调试进去发现这里的运算出现计算接收长度错误的情况。

[image]

mysterywolf, 2022-08-04:
https://github.com/RT-Thread/rt-thread/issues/6106

mysterywolf, 2022-08-04:
感谢反馈 已在此处修复 https://github.com/RT-Thread/rt-thread/pull/6226

closed 2022-08-05 by mysterywolf