RT-Thread/rt-thread #4557: rt_hw_us_delay 的实现

Guozhanxin, 2021-03-31:
stm32通用驱动如下：
```
/**
 * This function will delay for some us.
 *
 * @param us the delay time of us
 */
void rt_hw_us_delay(rt_uint32_t us)
{
    rt_uint32_t start, now, delta, reload, us_tick;
    start = SysTick->VAL;
    reload = SysTick->LOAD;
    us_tick = SystemCoreClock / 1000000UL;
    do
    {
        now = SysTick->VAL;
        delta = start >= now ? start - now : reload + start - now;
    }
    while (delta < us_tick * us);
}
```

文档里给的实例：
```
#include <board.h>
void rt_hw_us_delay(rt_uint32_t us)
{
    rt_uint32_t tstart;
    /* 获得延时经过的 tick 数 */
    us = us * (SysTick->LOAD/(1000000/RT_TICK_PER_SECOND));
    /* 获得当前时间 */
    tstart = SysTick->VAL;
    /* 判断是否会溢出 */
    if (tstart < us) tstart += SysTick->LOAD;
    /* 循环获得当前时间，直到达到指定的时间后退出循环 */
    while (tstart - SysTick->VAL < us);
}
```

大家感觉哪种更好一点呢？

Guozhanxin, 2021-03-31:
```
/**
 * The time delay function.
 *
 * @param microseconds.
 */
static void stm32_udelay(rt_uint32_t us)
{
    rt_uint32_t ticks;
    rt_uint32_t told, tnow, tcnt = 0;
    rt_uint32_t reload = SysTick->LOAD;

    ticks = us * reload / (1000000 / RT_TICK_PER_SECOND);
    told = SysTick->VAL;
    while (1)
    {
        tnow = SysTick->VAL;
        if (tnow != told)
        {
            if (tnow < told)
            {
                tcnt += told - tnow;
            }
            else
            {
                tcnt += reload - tnow + told;
            }
            told = tnow;
            if (tcnt >= ticks)
            {
                break;
            }
        }
    }
}
```

通用驱动 soft_i2c.c 里的

mysterywolf, 2021-04-18:
https://gitee.com/rtthread/rt-thread/issues/I26USG

mysterywolf, 2021-06-08:
https://github.com/RT-Thread/rt-thread/issues/4775

Guozhanxin, 2021-06-10:
https://gitee.com/rtthread/rt-thread/pulls/217

closed 2021-06-16 by mysterywolf