RT-Thread/rt-thread #4577: kernel panic: mktime backtrace问题

KelvinTsoi, 2021-04-07:
使用cmbacktrace工具dump出的信息，请问大家有头绪吗？使用的是latest stable-v3.0.x，谢谢。
```
0x20000638
$d
impure.c:?
0x0804d3e8
_findenv_r
??:?
0x0804d432
_getenv_r
??:?
0x0804c010
_tzset_unlocked_r
??:?
0x08049f1e
mktime
??:?
0x08048e6c
rt_rtc_control
/home/kevin/hook/ecu/pdu/RTTOS/bsp/stm32/libraries/HAL_Drivers/drv_rtc.c:148
0x0801ef92
rt_exit_critical
```

KelvinTsoi, 2021-04-07:
=================== Registers information ====================
  R0 : ffffffed  R1 : ffffffed  R2 : 00000002  R3 : 20000bac
  R12: 0805c434  LR : 20000b08  PC : 20000638  PSR: 2001ba1c
==============================================================
Bus fault is caused by precise data access violation
The bus fault occurred address is 0316960d
Show more call stack info by run: addr2line -e *.elf -a -f 20000638 0804d3e8 0804d432 0804c010 08049f1e 08048e6c 0801ef92 0801\0

mysterywolf, 2021-04-07:
用最新版的master试试 最近这部分改了一些bug和结构不合理的地方

closed 2021-04-17 by mysterywolf