RT-Thread/rt-thread #4610: [kernel] mem.c 上游同步问题

mysterywolf, 2021-04-13:
kernel mem.c貌似已经很久没有和LWIP的mem.c同步过了 https://github.com/lwip-tcpip/lwip/blob/master/src/core/mem.c
我随便翻了翻就发现了几处 LWIP后续修改的bug 但是kernel这边没有同步过来
后续需要仔细比对一下

xiangxistu, 2021-04-14:
看着一下，差异确实有一些。

xiangxistu, 2021-04-20:
@mysterwolf，满老师：

通过比对 lwip/core/mem.c 里的代码，主要是内存的分配，释放，与小内存的合并部分；我发现了有三处不太一致的地方，如果是逻辑部分的话，有 2 处；

1. lwip 提供的 ```mem_trim(void *rmem, mem_size_t new_size)``` 函数，应该只是 RT-Thread 的 rt_realloc 函数的一部分功能；即，mem_trim 只允许缩小已经申请的堆大小；而 rt_realloc 允许缩小或者扩大申请的堆大小（当然，扩大堆空间大小的方式是通过申请一个新的空间，拷贝的旧数据实现的）。
2. lwip 提供的函数 ```ptr_to_mem(mem_size_t ptr)``` 与 ```mem_to_ptr(void *mem)``` 实质上与 RT-Thread 的用法没有区别；但是对 ```plug_holes``` 的堆末尾的空的 mem 做了保护，RT-Thread 没有改变。
3. lwip 允许在申请空间的操作被打断，而 RT-Thread 不允许这种操作；lwip 实现的途径是基于查询内存堆是否被修改的 flag 实现的，在 malloc 中会释放互斥量并打开中断，之后会检验 ```mem_free_count``` 的值，一旦 flag 被修改（mem_free 与 mem_tirm都会变更这个状态）则再重头开始申请空间。

----

内存的检查部分的逻辑，还需要同步吗？

mysterywolf, 2021-04-20:
你好，因为现在rtt的mem.c已经和lwip的原生mem差异较大，肯定不能lwip mem改什么 rtt这头就改什么，建议跟踪一下每一个lwip mem的commit 评估一下每个commit对rtt mem.c的影响，如果已经和rtt mem意图不符，则此commit不应该并入rtt这头。我没有仔细看具体差异是什么，但是我留意到lwip mem增加了一些对条件的判断，如果对rtt有利，则应该合并进去。不过总之，rtt的mem应该已经有年头没有和lwip这边同步了，应该考虑同步一次了。

BernardXiong, 2021-04-21:
可以多考虑memheap的方式，这种的会更合理

mysterywolf, 2021-07-02:
https://github.com/RT-Thread/rt-thread/pull/4777
https://github.com/RT-Thread/rt-thread/pull/4783