RT-Thread/rt-thread #4625: tm4c129x drv_eth.c tivaif_receive bug

Simon-Leung, 2021-04-20:
[https://github.com/RT-Thread/rt-thread/blob/master/bsp/tm4c129x/drivers/drv_eth.c](url)

line 866

```
if((rt_mb_send(dev->rx_pbuf_mb, (rt_uint32_t)pBuf) != RT_EOK) ||
    (eth_device_ready(&(dev->parent)) != RT_EOK))
{
  /* drop the packet */
  LWIP_DEBUGF(NETIF_DEBUG, ("tivaif_input: input error\n"));
  pbuf_free(pBuf);

  /* Adjust the link statistics */
  LINK_STATS_INC(link.memerr);
  LINK_STATS_INC(link.drop);
  DRIVER_STATS_INC(RXPacketCBErrCount);
}
```

If `rt_mb_send(dev->rx_pbuf_mb, (rt_uint32_t)pBuf)` is OK, but `eth_device_ready(&(dev->parent)` is not ok, pBuf will be free.
The `eth_rx_thread_entry` will be using a free pBuf when next recv with both ok.

Simon-Leung, 2021-04-20:
and the line 835, pBuf dosn't assign to NULL after free.
The TivaWare sample was assign to NULL in this block.

```
/* Yes - does the frame contain errors? */
if(pDescList->pDescriptors[pDescList->ui32Read].Desc.ui32CtrlStatus &
   DES0_RX_STAT_ERR)
{
    /* This is a bad frame so discard it and update the relevant
     * statistics.
     */
    LWIP_DEBUGF(NETIF_DEBUG, ("tivaif_receive: packet error\n"));
    pbuf_free(pBuf);
    LINK_STATS_INC(link.drop);
    DRIVER_STATS_INC(RXPacketErrCount);
}
```

BernardXiong, 2021-04-21:
Yes,

It's better to do `tivaif_receive` in erx thread and not the isr. Welcome PR and I haven't tm4c129x board yet.

BernardXiong, 2021-04-21:
@ArdaFu Have you any comments? Thanks.

ArdaFu, 2021-06-25:
Hi @BernardXiong, here is the patches to fix the eth buffer free errors. I also upgraded the TI library. 
Since I didn't have a TM4C129x board in hand, maybe we can ask @Simon-Leung to help us testing the patched code.
Please DO NOT marge it until Simon finish the evaluation.

Simon-Leung, 2021-06-29:
Sorry, I don't have this bsp any more.

Simon-Leung, 2021-06-29:
Sorry, I don't have this board any more.

ArdaFu, 2021-06-29:
> Sorry, I don't have this board any more.

All right, I'll try to test it when I find one.

closed 2021-06-29 by ArdaFu