RT-Thread/rt-thread #4627: [bug] STM32 drv_rtc.c

mysterywolf, 2021-04-20:
https://club.rt-thread.org/ask/question/430893.html

mysterywolf, 2021-04-24:
这个issue也提到这个问题了https://github.com/RT-Thread/rt-thread/issues/3590

mysterywolf, 2021-05-17:
https://gitee.com/rtthread/rt-thread/issues/I3RJAO

mysterywolf, 2021-05-23:
https://github.com/RT-Thread-packages/netutils/issues/69

mysterywolf, 2021-05-26:
https://club.rt-thread.org/ask/question/427284.html
[image]

mysterywolf, 2021-06-02:
https://gitee.com/rtthread/rt-thread/issues/I3SEKR

CraztTnspt, 2021-06-03:
通过了解上面提到的问题

**发现以下两种导致 RTC 使用出现问题的原因：**

1. 仅通过 menuconfig 开启 RTC 功能，但未在 CubeMX 中作相应配置。
2. 受一个已修复 Bug1 影响的问题（在 [PR 3137](https://github.com/RT-Thread/rt-thread/pull/3137) 中引入，在 [PR 3591](https://github.com/RT-Thread/rt-thread/pull/3591) 中修复），持续时间从 2019-10-17 至 2020-05-12。

**上面提到的问题可大致分类如下：**

1. 由于未配置 CubeMX，导致错误。但我在F401RE 与 F103ZET6 上测试时，未在 CubeMx 中配置会造成编译报错，和下面论坛提问中的现象不相符，原因未知。

   - [stm32f103 bsp 里RTC 无法保存 年月日 的问题 找到原因 - RT-Thread问答社区](https://club.rt-thread.org/ask/question/430893.html)

   

2. 受 Bug1 影响

   - [rtc 重启后，时间重置问题 · Issue #3590 ](https://github.com/RT-Thread/rt-thread/issues/3590)（问题描述较少，从时间上看可能受 Bug1 影响）

   - [硬件RTC后背寄存器访问出错RTC hasn't been configured - RT-Thread问答社区](https://club.rt-thread.org/ask/question/427284.html) 。在此处也由提问者本人定位问题并给出了解决方法。这个论坛提问的时间是在 2020 年 9月，但却受到于 2020 年 5月被修复的 Bug1 影响，可能是使用了引入bug后还未修复的源码。

3. 未知原因：
   - [RTC 报“RTC hasn't been configured, please use <date> command to config.” 问题 · Issue #I3SEKR · RT-Thread/rt-thread - Gitee.com](https://gitee.com/rtthread/rt-thread/issues/I3SEKR) 在 F103ZET6 上未复现此问题，同时该 Issue 中提到的原因应该也不正确，`__HAL_RCC_BKP_CLK_ENABLE()` 实现在 `rt_rtc_f1_bkp_update()` 中，`rt_rtc_config()`通过调用`rt_rtc_f1_bkp_update()` 实现了时钟使能。实际错误原因可能需要在相同平台的工程上寻找？

iysheng, 2021-09-30:
我发现了一个问题在 stm32 的 bsp 中有大量的 RTC 配置在 board/board.c 文件中初始化为使用外部 LSE 使用作为 RTC 的时钟源.
[image]
这就会导致一个问题,当使用 menuconfig 选择了 LSI 作为 RTC 时钟,这时候就会导致在执行函数`HAL_RCCEx_PeriphCLKConfig(&PeriphClkInitStruct)` 是因为时钟选择的不匹配导致 reset backup 寄存器,进一步导致读取的 backup 寄存器的值为 0x0 而打印未配置时间的错误信息:
[image].

mysterywolf, 2021-10-13:
这个咋关了？问题还没有解决呢

mysterywolf, 2021-11-14:
https://club.rt-thread.org/ask/article/3143.html
论坛报告新的问题

mysterywolf, 2022-05-25:
目前 只剩下了 https://github.com/RT-Thread/rt-thread/pull/5116

closed 2022-05-31 by mysterywolf