RT-Thread/rt-thread #4635: BSP 中K210编译无法通过。

chunyexixiaoyu, 2021-04-23:
因为勘智的库里面的sleep函数和rt官方libc中的 unistd.c 重定义，以及rt_hw_us_delay函数无定义
[image]

mysterywolf, 2021-04-24:
你可以使用对方库里的sleep，或者注释掉对方库里的sleep，使用rtt的sleep，关于rt_hw_us_delay，这个需要用户自己定义，如果不想实现，可以给一个空函数，后续会有一个pr合并，替用户实现这个空函数，以免报错。

chunyexixiaoyu, 2021-04-25:
> 你可以使用对方库里的sleep，或者注释掉对方库里的sleep，使用rtt的sleep，关于rt_hw_us_delay，这个需要用户自己定义，如果不想实现，可以给一个空函数，后续会有一个pr合并，替用户实现这个空函数，以免报错。#4510

我就是把这个问题提出来

BernardXiong, 2021-04-25:
或者RTT里的sleep设置成weak吧。rt_hw_us_delay不知道什么梗

chunyexixiaoyu, 2021-05-07:
``/* Copyright 2018 Canaan Inc.
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */
#include "sleep.h"
#include "sysctl.h"

int usleep(uint64_t usec)
{
    uint64_t cycle = read_cycle();
    uint64_t nop_all = usec * sysctl_clock_get_freq(SYSCTL_CLOCK_CPU) / 1000000UL;
    while (1)
    {
        if(read_cycle() - cycle >= nop_all)
            break;
    }
    return 0;
}

int msleep(uint64_t msec)
{
    return (unsigned int)usleep(msec * 1000);
}

unsigned int sleep(unsigned int seconds)
{
    return (unsigned int)msleep(seconds * 1000);
}

chunyexixiaoyu, 2021-05-07:
@mysterywolf  
[image]

Guozhanxin, 2021-05-08:
> 或者RTT里的sleep设置成weak吧。rt_hw_us_delay不知道什么梗

我们的sleep会让出处理器，如果被其他第三方库覆盖掉之后，会不会容易导致其他的问题。。。

Guozhanxin, 2021-05-08:
我是感觉有些函数，有冲突，报错就报错了。如果没有解决问题，而是把问题隐藏起来了，后面就更麻烦了。。。

chunyexixiaoyu, 2021-05-08:
@mysterywolf  我觉得加上弱引用不影响用户选择，对于K210 BSP来说已经有申明的sleep肯定会调用它吧。可能我理解的片面。

mysterywolf, 2021-05-08:
@chunyexixiaoyu 我觉得郭老师说的是对的，这个SDK貌似没有和操作系统深度融合，这个sleep采用的是死卡的方式，正常应该采用调度的方式进行，sleep是秒级别的延时，一卡卡好几秒这谁受得了。

chunyexixiaoyu, 2021-05-08:
@mysterywolf  懂了，可是 sdk是另外的仓库了，有没有比较优雅的方法能让用户直接可以编译不报错

mysterywolf, 2021-05-08:
这个应该不是rtt的问题了，因为是sdk的sleep设计不当导致的，sdk的sleep死卡1秒钟这个既不符合linux规范，也不符合操作系统设计思想。应该去sdk那头提issue

chunyexixiaoyu, 2021-05-08:
> 这个应该不是rtt的问题了，因为是sdk的sleep设计不当导致的，sdk的sleep死卡1秒钟这个既不符合linux规范，也不符合操作系统设计思想。应该去sdk那头提issue。我们这头无法解决由于对方设计问题导致的错误。

我觉得这个问题可以写在readme里面，sdk是K210的官方的支持包，是针对裸跑的，他的延时就是硬卡，对于裸跑来说也没毛病的。

mysterywolf, 2021-05-08:
sdk是裸机的啊，那是需要适配的或者说移植的，直接落到rt-thread上还要求不报错，有些强人所难。写到readme这个，因为rt-thread的bsp并没有含着sdk，所以没有必要写到readme里边。

chunyexixiaoyu, 2021-06-04:
@BernardXiong @mysterywolf @Guozhanxin   
发现了一个新问题，这个不能编译是存在的，解决这不能编译的问题发现如果在k210 sdk sleep函数加上weak 导致整个系统会起不来，因为board.c 文件中初始化中
`void rt_hw_board_init(void)
{
    
    sysctl_pll_set_freq(SYSCTL_PLL0, 800000000UL);
    sysctl_pll_set_freq(SYSCTL_PLL1, 400000000UL);
    /* Init FPIOA */
    fpioa_init();
    /* Dmac init */
    dmac_init();

    /* initalize interrupt */
    rt_hw_interrupt_init();
    /* initialize hardware interrupt */
    rt_hw_uart_init();

    rt_hw_tick_init();

#ifdef RT_USING_SMP
    rt_hw_clint_ipi_enable();
#endif

#ifdef RT_USING_CONSOLE
    /* set console device */
    rt_console_set_device(RT_CONSOLE_DEVICE_NAME);
#endif /* RT_USING_CONSOLE */

#ifdef RT_USING_HEAP
    rt_kprintf("heap: [0x%08x - 0x%08x]\n", (rt_ubase_t) RT_HW_HEAP_BEGIN, (rt_ubase_t) RT_HW_HEAP_END);
    /* initialize memory system */
    rt_system_heap_init(RT_HW_HEAP_BEGIN, RT_HW_HEAP_END);
#endif

#ifdef RT_USING_COMPONENTS_INIT
    rt_components_board_init();
#endif
}`
sysctl_pll_set_freq函数会用到usleep，但是这个时候定时器中断还没有设置，然后会死在
[image]

但是如果用posix接口做应用的话，目前看来只能改k210  sdk sleep 函数了。

mysterywolf, 2021-06-15:
该问题 已经解决 https://github.com/RT-Thread/rt-thread/pull/4797

closed 2021-06-15 by mysterywolf

chunyexixiaoyu, 2021-06-16:
> 该问题 已经解决 #4797

那这个前提条件是如果用posix接口的话，rt-thread fork勘智的那个sdk那个库 这两个函数应该加_weak了吧