RT-Thread/rt-thread #4704: [TODO] 所有的RT_IPC_FLAG_FIFO改为RT_IPC_FLAG_PRIO

mysterywolf, 2021-05-17:
不管是邮箱、消息队列、互斥量还是信号量，所有的RT_IPC_FLAG_FIFO改为RT_IPC_FLAG_PRIO；RT_IPC_FLAG_FIFO并不是实时操作系统的一部分，如果采用RT_IPC_FLAG_FIFO初始化，将会导致系统变为非实时。详见：https://github.com/RT-Thread/rt-thread/pull/4703

Guozhanxin, 2021-05-18:
#### 正常思路
1. 要看有没有哪种实际需求，是要让线程按照FIFO的顺序排序等待的。
2. 在进行排序等待的过程中，等待队列中的线程此时是否仍然关注实时性？

#### 异常思路
1. 如果按照FIFO的方式排队的线程就不太关注实时性了，那么会导致的最严重后果是什么
2. 如果按照FIFO的方式排队的线程仍然关注实时性，**进行FIFO排序只是这个资源本身的问题，不得已才按照FIFO排序**，导致的最严重后果是什么

mysterywolf, 2021-05-18:
FIFO 和实时性是高冲突概念。二者不可调和，按照实时性排序就不能按照FIFO排序，按照FIFO排序就不能按照先来后到排序。我们是实时操作系统，FIFO不属于实时操作系统的一部分，只算是一个广义OS的一部分，属于一个附加项，作为一个实时操作系统，我们的所有组件却采用的非实时的方式，这是不能容忍的。

mysterywolf, 2021-07-07:
https://github.com/RT-Thread/rt-thread/pull/4852

closed 2021-07-07 by mysterywolf

mysterywolf, 2021-07-28:
https://gitee.com/rtthread/docs-online/pulls/154

mysterywolf, 2021-07-28:
https://github.com/RT-Thread/rt-thread/pull/4875

mysterywolf, 2021-07-28:
https://github.com/RT-Thread-packages/kernel-sample/pull/22