RT-Thread/rt-thread #4724: BSP NUC980 boot from SPI NAND Operate file return [PDMA-W]

studyinhub, 2021-05-23:
Boot from SPI Nand,when I mkdir in /mnt/filesystem , it return [PDMA-W].But if I run from DDR,it's mostly OK,what's the problem?Can anyone help?

Here is the logs:

U-Boot 2016.11 (Nov 04 2020 - 19:38:19 +0800)

CPU: NUC980
Board: NUC980
DRAM:  64 MiB
NAND:  NAND Flash not found !
NUC980 NAND CONTROLLER IS NOT SUPPORT THE PAGE SIZE. (0, 0)
0 MiB
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
SF: Detected W25N01GV with page size 2 KiB, erase size 128 KiB, total 128 MiB
device 0 offset 0x200000, size 0x100000
SF: 1048576 bytes @ 0x200000 Read: OK
## Starting application at 0x00000000 ...

 \ | /
- RT -     Thread Operating System
 / | \     4.0.4 build May 24 2021
 2006 - 2021 Copyright by rt-thread team
lwIP-2.0.2 initialized!
[I/I2C] I2C bus [i2c0] registered
[I/I2C] I2C bus [i2c2] registered
[I/spinand_flash] Found: [00EFAA21] Winbond 128MB: 2048+64@64@1024.
[I/spinand_flash] Enabled BUF, HWECC. Unprotected.
[I/sal.skt] Socket Abstraction Layer initialize success.
[I/utest] utest is initialize success.
[I/utest] total utest testcase num: (0)
os  : system memory alloc 320 bytes
flsh: UFFS consume spare data size 49
os  : system memory alloc 83520 bytes
os  : system memory alloc 52400 bytes
os  : system memory alloc 16000 bytes
tree: DIR 0, FILE 0, DATA 0
[I/wn] RT-Thread webnet package (V2.0.3) initialize success.
MAC address: 82:06:21:53:94:53
nu_igmp_mac_filter add 224.0.0.1 01:00:5E:00:00:01
Find uart8.
Set uart8 to RS-485 AUD function mode. ActiveLowLevel-NO
[D/wn.log] server initialize success.
msh />
msh />l[e0] Link status: 100F
s
Directory /:
mnt                 <DIR>
cache               <DIR>
download            <DIR>
msh />cd /mn
msh />cd /mnt
msh /mnt>ls
Directory /mnt:
ram_usbd            <DIR>
filesystem          <DIR>
msh /mnt>mkdir nand1
msh /mnt>ls
Directory /mnt:
ram_usbd            <DIR>
filesystem          <DIR>
nand1               <DIR>
msh /mnt>mount
filesystem  device  mountpoint
----------  ------  ----------
devfs       (NULL)  /dev
elm         ramdis  /
elm         ramdis  /mnt/ram_usbd
uffs        nand1   /mnt/filesystem
msh /mnt>tftp -s
tftp server start!
msh /mnt>[PDMA-W]

studyinhub, 2021-05-23:
@wosayttn  Could you give a hand?

wosayttn, 2021-05-24:
Hi, @studyinhub 

Here, the [PDMA-W](https://github.com/RT-Thread/rt-thread/blob/master/bsp/nuvoton/libraries/nuc980/rtt_port/drv_pdma.c#L690)'s meaning is as below:
```bash
   /*
       Race-condition avoidance between DMA-transferring and DCache write-back:
       Source, destination, DMA descriptor address and length should be aligned at len(CACHE_LINE_SIZE).
   */
```

wosayttn, 2021-05-24:
PDMA driver 內並無實現一 temp DMA buffer 緩存所有的 request DMA transation 資料。
為避免快取資訊一致性問題，這邊為進行偵測且提醒，可能發生的 race-condition 問題。

所以，DMA 用的 buffer size 和 start address 要在 cache line 的 boundary 上，buffer 長度也是 cache line 的倍數。

closed 2021-11-20 by studyinhub