RT-Thread/rt-thread #4728: M0没有SCB_VTOR 不能通过此寄存器恢复MSP

WUSY1991, 2021-05-25:
rt_hw_context_switch_to 函数 应判断从SCB_VTOR读出来的值，为0不要去修改Msp，否则后面的代码会执行失败
（M0不是所有的core 都有SCB_VTOR的）

; restore MSP
    LDR     r0, =SCB_VTOR
    LDR     r0, [r0]
    LDR     r0, [r0]
    MSR     msp, r0

aozima, 2021-05-25:
赞同

有较好的办法区分当前芯片的SCB_VTOR是否有效吗？
不然针对M0只能删除这部分代码。

WUSY1991, 2021-05-25:
VTOR 目前一般不会是0，或者做一次读写操作确认

aozima, 2021-06-09:
我是这样考虑的

根据ARM的规定，SRAM地址在`0x20000000~0x3FFFFFFF`，所以MSP肯定是在这个范围内。
且栈地址要求8字节对齐。
因此可以得到判断条件掩码 0x20000007，判断MSP是否合法。

```
0x20000007 // 0010 0000 0000 0000 0000 0000 0000 0111
0x20000000 // 0010 0000 0000 0000 0000 0000 0000 0000
0x3FFFFFFF // 0011 1111 1111 1111 1111 1111 1111 1111
```

改好之后是这样，麻烦有条件帮实际验证下
```asm
    ; restore MSP. (0x20000000 <= MSP <= 0x3FFFFFFF), 8-byte aligned. 
    LDR     r0, =SCB_VTOR
    LDR     r0, [r0]
    LDR     r0, [r0]
    LDR     r1, =0x20000007
    ANDS    r1, r1, r0
    CBNZ    r0, restore_msp_end
    MSR     msp, r0
restore_msp_end
```

aozima, 2023-03-24:
>判断MSP是否合法。

这里的思路应该是没问题的，但判断应该改进得更严谨些，可能 异或 xor 运算更好些。

cms19859230182-lang, 2026-11-12:
已在 master (`611d7c07d5dd32757ac6d363576e5f0e8b915c3c`) 上确认三份 Cortex-M0 `context_{gcc,rvds,iar}.S` 的 `rt_hw_context_switch_to` 仍无条件 `MSR msp, r0`。修复 PR：#12044。在读取 `SCB_VTOR` 后先判 0，为 0 时跳过 MSP 恢复，非 0 路径保持原样；本 PR 只改 VTOR 段，不包含 HardFault 修改。

closed 2026-11-13 by Rbb666