RT-Thread/rt-thread #4755: 关于arm-a系列启动代码的问题

huyu168, 2021-06-02:
目前我用的是一块树莓派4开发板，最近一位大佬更新了一段启动代码，于是我试图开启smp，但是我发现在唤醒其他副核时无法唤醒，唤醒的方法是将启动代码的地址写入指定内存，最后发现下面这一段start_gcc中，本应该循环等待的副核并没有进入循环。我想问一下a72的启动时，其余副核不会和主核一起运行吗？还是说他们不会走_rest（这段循环的函数）吗？
`ifdef RT_SMP_AUTO_BOOT
    ldr r0, =secondary_cpu_entry
    mov r1, #0
    str r1, [r0] /* clean secondary_cpu_entry */
#endif /* RT_SMP_AUTO_BOOT */

secondary_loop:
    @ cpu core 1 goes into sleep until core 0 wakeup it
    wfe
#ifdef RT_SMP_AUTO_BOOT
    ldr r1, =secondary_cpu_entry
    ldr r0, [r1]
    cmp r0, #0
    blxne r0 /* if(secondary_cpu_entry) secondary_cpu_entry(); */
#endif /* RT_SMP_AUTO_BOOT */
    b secondary_loop
`

huyu168, 2021-06-02:
@zhouji92 希望大佬能帮忙解答一下

zhouji92, 2021-06-02:
* 不同的芯片，副核启动方式可能不一样，主要看电源和时钟这块。一般分两种
1. 所有核心一起启动，这种不常见，有些国产厂家喜欢这样干，所以启动代码中会根据cpu id来判断是不是cpuid 0，不是cpu 0的会进入secondary_loop。
2. 主核心(`cpuid == 0`)启动，然后由主核唤醒附属核心，大多数芯片是这样的，比如三星s5p6818 副核启动前要选择运行模式，设置启动地址，详情请看手册第257页。

zhouji92, 2021-06-02:
> @zhouji92 希望大佬能帮忙解答一下

* 不同的芯片，副核启动方式可能不一样，主要看电源和时钟这块。一般分两种
1. 所有核心一起启动，这种不常见，有些国产厂家喜欢这样干，所以启动代码中会根据cpu id来判断是不是cpuid 0，不是cpu 0的会进入secondary_loop。
2. 主核心(`cpuid == 0`)启动，然后由主核唤醒附属核心，大多数芯片是这样的，比如三星s5p6818 副核启动前要选择运行模式，设置启动地址，详情请看手册第275页。

zhouji92, 2021-06-03:
@huyu168 兄弟你不点个赞吗。

huyu168, 2021-06-23:
> @huyu168 兄弟你不点个赞吗。
感谢！休假了一段时间，没有看github了

closed 2021-06-23 by mysterywolf