RT-Thread/rt-thread #4763: STM32 drv_common.c fails to start with CubeMX

majianjia, 2021-06-04:
The initialisation sequence is incorrect. 

~~~
    /* HAL_Init() function is called at the beginning of the program */
    HAL_Init();  ---> it sometime requires tick support and Systick interrupt.

    /* enable interrupt */
    __set_PRIMASK(0);
    /* System clock initialization */
    SystemClock_Config();
    /* disable interrupt */
    __set_PRIMASK(1);

    rt_hw_systick_init(); --->tick support inited. 
~~~

In a fully configurable CubeMX's `HAL_Init()`, it requires the tick support, thus it needs Systick to interrupt.  The current sequence only work with some simple CubeMX setting that has empty or non-waited `HAL_MspInit();` (called by `HAL_Init()`)

This `HAL_MspInit();` requires the tick for initialisation, generated by CubeMX. 

~~~
void HAL_MspInit(void)
{
  /* USER CODE BEGIN MspInit 0 */

  /* USER CODE END MspInit 0 */

  __HAL_RCC_SYSCFG_CLK_ENABLE();

  /* System interrupt init*/

  /** Enable the VREF clock
  */
  __HAL_RCC_VREF_CLK_ENABLE(); --->> does not required tick
  /** Configure the internal voltage reference buffer voltage scale
  */
  HAL_SYSCFG_VREFBUF_VoltageScalingConfig(SYSCFG_VREFBUF_VOLTAGE_SCALE0);
  /** Enable the Internal Voltage Reference buffer
  */
  HAL_SYSCFG_EnableVREFBUF();  --->>> This method required tick to wait for stable.
  /** Configure the internal voltage reference buffer high impedance mode
  */
  HAL_SYSCFG_VREFBUF_HighImpedanceConfig(SYSCFG_VREFBUF_HIGH_IMPEDANCE_DISABLE); --->>> This method also require ticks. 

  /* USER CODE BEGIN MspInit 1 */

  /* USER CODE END MspInit 1 */
}
~~~
In a simple MspInit(), there is only `__HAL_RCC_VREF_CLK_ENABLE(); ` so all previous STM32 bsp works. 
But here I add the configuration of Vrefbuf, which require ticks to wait for stability, then it stuck because the tick never changed.

thewon86, 2021-07-15:
https://club.rt-thread.org/ask/article/2881.html

majianjia, 2021-07-15:
@thewon86 
Thanks but your proposal cannot solve this issue.

thewon86, 2021-07-19:
> @thewon86
> Thanks but your proposal cannot solve this issue.

show your reason and explanation please.

majianjia, 2021-07-19:
As mentioned in the op, HAL_Init() in some configuration needs rt tick running. Simply block interrupt does change the init sequence, therefore still block the HAL_Init()

thewon86, 2021-07-19:
> As mentioned in the op, HAL_Init() in some configuration needs rt tick running. Simply block interrupt does change the init sequence, therefore still block the HAL_Init()

As mentioned in the link, you can replace tick by a combination of while and udelay. no interrupt, no tick, is ok.
see the example in it 
```
    /* HAL_Init() function is called at the beginning of the program */
    HAL_Init();

    /* System clock initialization */
    SystemClock_Config();

    rt_hw_systick_init();
```

no interruption enabled.