RT-Thread/rt-thread #4791: [TODO] 解决RT_Exxx错误代码以及POSIX错误代码正负乱用的问题

mysterywolf, 2021-06-12:


Guozhanxin, 2021-06-13:
这个问题其实不止是正负乱用，有些组件错误码本身也没有区分清楚。

mysterywolf, 2021-06-13:
https://github.com/RT-Thread/rt-thread/issues/2002

mysterywolf, 2021-06-13:
建议我们这边得做出硬性规定，用RTT自己的错误代码用负数表示，Linux/posix错误代码跟随linux用正数

mysterywolf, 2021-06-17:
https://gitee.com/rtthread/rt-thread/issues/I3VR7D#note_5482613

Guozhanxin, 2021-06-18:
| RT ERROR    | 描述                                                         | posix   |
| ------- | ------------------------------------------------------------ |-------------- |
| 00 RT_EOK     | no error                                           |  0 |
| 01 RT_ERROR      | generic error                             |  |
| 02 RT_ETIMEOUT      | Timed out            |  ETIMEDOUT 60|
| 03 RT_EFULL     | The resource is full                     |    |
| 04 RT_EEMPTY    | The resource is empty            |  |
| 05 RT_ENOMEM  | No memory                              | ENOMEM 12 |
| 06 RT_ENOSYS      | No system                                        | ENOSYS 88 |
| 07 RT_EBUSY  | Busy                                                |  EBUSY 16|
| 08 RT_EIO | IO error                                                |  EIO 5 |
| 09 RT_EINTR      | Interrupted system call             | EINTR 4 |
| 10 RT_EINVAL      | Invalid argument                              | EINVAL 22 |

Guozhanxin, 2021-06-18:
需要注意文件抽象层 dfs 和网络抽象层 sal 的存在，会不会对 posix error 的返回造成影响。

closed 2023-10-23 by mysterywolf