RT-Thread/rt-thread #4794: [建议] STM32 HAL库可否直接以submodule方式引用官方的仓库？

mysterywolf, 2021-06-13:
这样可以随时保持和官方最新的release版本保持同步，维护也比较方便，用户也可以用最新版的hal库，毕竟hal内部还是有些bug的。

https://github.com/STMicroelectronics/stm32f1xx_hal_driver
https://github.com/STMicroelectronics/stm32f2xx_hal_driver
https://github.com/STMicroelectronics/stm32f0xx_hal_driver
https://github.com/STMicroelectronics/stm32f4xx_hal_driver
https://github.com/STMicroelectronics/stm32h7xx_hal_driver
https://github.com/STMicroelectronics/stm32wbxx_hal_driver
https://github.com/STMicroelectronics/stm32l4xx_hal_driver
https://github.com/STMicroelectronics/stm32f7xx_hal_driver
https://github.com/STMicroelectronics/stm32l5xx_hal_driver
https://github.com/STMicroelectronics/stm32g4xx_hal_driver
https://github.com/STMicroelectronics/stm32l0xx_hal_driver
https://github.com/STMicroelectronics/stm32f3xx_hal_driver
https://github.com/STMicroelectronics/stm32wlxx_hal_driver

loogg, 2021-06-13:
我也同意，库和驱动放在另一个仓库里。
最好bsp也不要放在源码中。

mysterywolf, 2021-06-13:
另外建议考虑各个芯片厂商的官方HAL库应该单独放在一个仓库里，让芯片厂商自己维护，针对rtt操作系统drv_xxx文件可以放在bsp里边，就像软件包一样，用的时候临时从官方的仓库里边拉取，这样维护HAL库的任务就交给芯片厂商就可以了

Guozhanxin, 2021-06-16:
也要考虑新手入门时应该怎么搞，弄成 submodule 门槛就又高了一点

loogg, 2021-06-16:
1. 只要文档里写清楚就没事我觉得
2. 我觉得库的版本要指定好，不然可能出现Cube生成的是更高版本的库，而驱动呢比较老，导致两者不兼容出现各种问题

BernardXiong, 2021-07-02:
不建议用太多的submodule，是魔鬼

closed 2021-07-02 by BernardXiong