RT-Thread/rt-thread #4838: STM32 drv_usart.c ，uart1的rx、tx均使用DMA传输。会有一定概率出现tx无数据的情况

ynkan, 2021-07-01:
复现方法：3.1.5版本。在uart均tx、rx均使用DMA的情况下，正常传输中将rx接地(用杜邦线接地并抖动...)。此时可能会报LBD、FE标志。此后 tx_dma->activated 就会一直处于 true `状态。导致发送数据无法发送。

另外注释 以下语句可解决问题。但因为涉及型号众多。希望官方多型号测试。便于更好修复该问题。
....
else if (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_TC) != RESET &&
            (__HAL_UART_GET_IT_SOURCE(&(uart->handle), UART_IT_TC) != RESET))
{
        if ((serial->parent.open_flag & RT_DEVICE_FLAG_DMA_TX) != 0)
        {
            HAL_UART_IRQHandler(&(uart->handle));
        }
        //UART_INSTANCE_CLEAR_FUNCTION(&(uart->handle), UART_FLAG_TC);   //注释该句可解决问题。
}
....

该问题在实际项目中如果使用CP2102。CP2102在USB驱动识别的过程中CP2102-TX的电压可能会下降到2.7（vcc=3.3）并且一段时间内不稳定。就会导致上述现象。但此时STM32接收数据正常。

上述使用主控为STM32F405RGT6

KyleChenjh, 2021-07-01:
1. 增加抗干扰能力：软件上 检测下是不是串口没配置上拉，好像官网默认用cubemx配置的都是浮空。硬件上  也可以增加上拉提升抗干扰性能。
2. 中断处理部分：可以试试这样，先检测你遇到的错误，先清除一下错误标志再进行继续操作或者直接返回（下面示例是清除错误标志后继续执行）。
```c
static void uart_isr(struct rt_serial_device *serial)
{
    struct stm32_uart *uart;

    RT_ASSERT(serial != RT_NULL);
    uart = rt_container_of(serial, struct stm32_uart, serial);
    if (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_ORE) != RESET)
    {
        LOG_E("(%s) serial device Overrun error!", serial->parent.parent.name);
        __HAL_UART_CLEAR_OREFLAG(&uart->handle);
    }
    if (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_NE) != RESET)
    {
        LOG_E("(%s) serial device Noise error!", serial->parent.parent.name);
        __HAL_UART_CLEAR_NEFLAG(&uart->handle);
    }
    if (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_FE) != RESET)
    {
        LOG_E("(%s) serial device Framing error!", serial->parent.parent.name);
        __HAL_UART_CLEAR_FEFLAG(&uart->handle);
    }
    if (__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_PE) != RESET)
    {
        LOG_E("(%s) serial device Parity error!", serial->parent.parent.name);
        __HAL_UART_CLEAR_PEFLAG(&uart->handle);
    }

    if ((__HAL_UART_GET_FLAG(&(uart->handle), UART_FLAG_RXNE) != RESET) &&
            (__HAL_UART_GET_IT_SOURCE(&(uart->handle), UART_IT_RXNE) != RESET))
    {
... ...
```

arminker, 2021-07-20:
发送和接收采用dma没问题，但是貌似中断发送方式，实际上采用的是查询发送

Guozhanxin, 2021-07-20:
> 发送和接收采用dma没问题，但是貌似中断发送方式，实际上采用的是查询发送
可以看下 serialv2 https://github.com/RT-Thread/rt-thread/blob/master/components/drivers/serial/serial_v2.c

mysterywolf, 2021-07-27:
https://github.com/RT-Thread/rt-thread/pull/4851