RT-Thread/rt-thread #4844: 如何分别给 c代码和 C++代码设定编译参数

Guozhanxin, 2021-07-03:
按照下面这种写法，先把 CFlags 赋值给 CXXFlags，然后再单独给 CFlag 添加参数。编译时，新添加的参数仍然会传递给 CXX 编译器。
bsp: stm32-f429-apollo
编译器：armclang.
```
elif PLATFORM == 'armclang':
    # toolchains
    CC = 'armclang'
    CXX = 'armclang'
    AS = 'armasm'
    AR = 'armar'
    LINK = 'armlink'
    TARGET_EXT = 'axf'

    DEVICE = ' --cpu Cortex-M4.fp '
    CFLAGS = ' --target=arm-arm-none-eabi -mcpu=cortex-m4 '
    CFLAGS += ' -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 '
    CFLAGS += ' -mfloat-abi=hard -c -fno-rtti -funsigned-char -fshort-enums -fshort-wchar '
    CFLAGS += ' -gdwarf-3 -ffunction-sections '
    AFLAGS = DEVICE + ' --apcs=interwork '
    LFLAGS = DEVICE + ' --info sizes --info totals --info unused --info veneers '
   ...
    CXXFLAGS = CFLAGS
    CFLAGS += ' -std=c99'

    POST_ACTION = 'fromelf --bin $TARGET --output rtthread.bin \nfromelf -z $TARGET'
```

thewon86, 2021-07-05:
把特别的部分分离成库，多项目协作

Guozhanxin, 2021-07-06:
> ```python
> CXXFLAGS 
> ```

如果 CCFLAGS 是C和C++公共的参数，那 CFLAGS 和 CXXFLAGS 就应该**仅**包含各自独立的参数了？比如 ' -xc' 和 ' -xc++'

Guozhanxin, 2021-07-17:
https://github.com/RT-Thread/rt-thread/pull/2855

mysterywolf, 2021-09-05:
https://github.com/RT-Thread/rt-thread/pull/4953

liukangcc, 2021-10-14:
https://github.com/RT-Thread/rt-thread/pull/5171

mysterywolf, 2021-12-18:
https://github.com/RT-Thread/rt-thread/pull/5397

mysterywolf, 2022-01-21:
https://github.com/RT-Thread/rt-thread/pull/5411

closed 2022-01-21 by mysterywolf