RT-Thread/rt-thread #4847: 编译报错：undefined reference to usleep

songzeceng, 2021-07-05:
芯片：k210
工具链：xpack-riscv-none-embed-gcc-10.1.0-1.1-linux-x64
RT-Thread版本：当前master版本

复现步骤：
按照[说明](https://github.com/RT-Thread/rt-thread/tree/master/bsp/k210)中说的，先修改bsp/k210/rtconfig.py中的CPU和EXEC_PATH为k210和工具链的bin目录，再在bsp/k210目录下执行
```
source ~/.env/env.sh

cd bsp/k210
pkgs --update
scons
```
结果编译报错:
```
/home/rtlab/szc/riscv-none-embed-gcc-10.1.0-x64/bin/../lib/gcc/riscv-none-embed/10.1.0/../../../../riscv-none-embed/bin/ld: sysctl.c:(.text.sysctl_pll_set_freq+0x92): undefined reference to 'usleep'
collect2: error: ld returned 1 exit status
```
我检查了一下，sysctl_pll_set_freq()函数所在的bsp/k210/packages/kendryte-sdk-latest/kendryte-standalone-sdk/lib/drivers/sysctl.c包含的头文件bsp.h中已经包含了usleep()函数所在的头文件sleep.h，而我在rtconfig.py中给CFLAGS添加了sleep.h所在的头文件目录，还是不行，我感到费解。
最后，附上我的bsp/k210/rtconfig.py:
```
import os

# toolchains options
ARCH        ='risc-v'
CPU         ='k210'
CROSS_TOOL  ='gcc'

if os.getenv('RTT_ROOT'):
    RTT_ROOT = os.getenv('RTT_ROOT')
else:
    RTT_ROOT = r'../..'

if os.getenv('RTT_CC'):
    CROSS_TOOL = os.getenv('RTT_CC')

if  CROSS_TOOL == 'gcc':
    PLATFORM    = 'gcc'
    EXEC_PATH   = r'/home/rtlab/szc/riscv-none-embed-gcc-10.1.0-x64/bin'
else:
    print('Please make sure your toolchains is GNU GCC!')
    exit(0)

if os.getenv('RTT_EXEC_PATH'):
    EXEC_PATH = os.getenv('RTT_EXEC_PATH')

BUILD = 'release'

if PLATFORM == 'gcc':
    # toolchains
    PREFIX  = 'riscv-none-embed-'
    CC      = PREFIX + 'gcc'
    CXX     = PREFIX + 'g++'
    AS      = PREFIX + 'gcc'
    AR      = PREFIX + 'ar'
    LINK    = PREFIX + 'gcc'
    TARGET_EXT = 'elf'
    SIZE    = PREFIX + 'size'
    OBJDUMP = PREFIX + 'objdump'
    OBJCPY  = PREFIX + 'objcopy'

    DEVICE  = ' -mcmodel=medany -march=rv64imafdc -mabi=lp64d'
    CFLAGS  = DEVICE + ' -fno-common -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -L/usr/aarch64-linux-gnu/lib/'
    AFLAGS  = ' -c' + DEVICE + ' -x assembler-with-cpp'
    LFLAGS  = DEVICE + ' -nostartfiles -Wl,--gc-sections,-Map=rtthread.map,-cref,-u,_start -T link.lds'
    CPATH   = ''
    LPATH   = ''

    CFLAGS += " -I/home/rtlab/szc/rt-thread/bsp/k210/packages/kendryte-sdk-latest/kendryte-standalone-sdk/lib/bsp/include "

    if BUILD == 'debug':
        CFLAGS += ' -O0 -ggdb'
        AFLAGS += ' -ggdb'
    else:
        CFLAGS += ' -O2 -Os'

    CXXFLAGS = CFLAGS + " -std=c++17"

DUMP_ACTION = OBJDUMP + ' -D -S $TARGET > rtt.asm\n'
POST_ACTION = OBJCPY + ' -O binary $TARGET rtthread.bin\n' + SIZE + ' $TARGET \n'
```

另外，我再分享一下一个和工具链中结构体命名重复的问题，问题报错就是编译时说include/libc/libc_signal.h中的sigval、sigevent、siginfo和siginfo_t都在riscv-none-embed-gcc-10.1.0-x64/riscv-none-embed/include/sys/signal.h中预先定义了，我的解决方法是将rt-thread中涉及sigval、sigevent、siginfo和siginfo_t的全部改名为sigval_rt、sigevent_rt和siginfo_rt，siginfo_t也改成siginfo_rt_t，比如在include/libc/libc_signal.h中相关代码就成了如下所示：
```
#ifndef HAVE_SIGVAL
/*  Signal Generation and Delivery, P1003.1b-1993, p. 63
    NOTE: P1003.1c/D10, p. 34 adds sigev_notify_function and
          sigev_notify_attributes to the sigevent structure.  */

union sigval_rt
{
    int    sival_int;    /* Integer signal value */
    void  *sival_ptr;    /* Pointer signal value */
};
#endif

#ifndef HAVE_SIGEVENT
struct sigevent_rt
{
    int             sigev_notify;               /* Notification type */
    int             sigev_signo;                /* Signal number */
    union sigval_rt sigev_value;                /* Signal value */
    void            (*sigev_notify_function)( union sigval_rt );
                                             /* Notification function */
    void            *sigev_notify_attributes;   /* Notification Attributes, really pthread_attr_t */
};
#endif

#ifndef HAVE_SIGINFO
struct siginfo_rt
{
    rt_uint16_t si_signo;
    rt_uint16_t si_code;

    union sigval_rt si_value;
};
typedef struct siginfo_rt siginfo_rt_t;
#endif
```
自己再把rt-thread中涉及这几个结构体的地方改下名字即可，工作量不大

BernardXiong, 2021-07-05:
usleep应该是某个地方用到了，但是没有具体的实现。

后面这个结构体名称的问题，应该`scons -c`后，再`scons`可以解决吧。如果不是，可以后续试试怎么回事。

songzeceng, 2021-07-06:
> usleep应该是某个地方用到了，但是没有具体的实现。
> 
> 后面这个结构体名称的问题，应该`scons -c`后，再`scons`可以解决吧。如果不是，可以后续试试怎么回事。

是的，起码在sysctl_pll_set_freq()里用到了，但是usleep()也有实现，存在于sleep.c中，和被sysctl_pll_set_freq()所在的sysctl.c包含的头文件sleep.h对应。

结构体名称问题我已经解决了，方法就是给rt-thread中的结构体改名，改一个和工具链里的不冲突的即可

mysterywolf, 2021-07-06:
请删除k210里边自己带的sleep.c
rtthread 提供了usleep相关函数

mysterywolf, 2021-07-06:
https://github.com/RT-Thread-packages/kendryte_sdk
https://github.com/RT-Thread-packages/kendryte_sdk/pull/7

songzeceng, 2021-07-06:
有么，我全局搜了一下，usleep()的实现只有k210和和bsp/gkipc/armv6/rtos_lib.c中有

songzeceng, 2021-07-06:
> https://github.com/RT-Thread-packages/kendryte_sdk
> [RT-Thread-packages/kendryte_sdk#7](https://github.com/RT-Thread-packages/kendryte_sdk/pull/7)

而且我把k210里的sleep.c重命名为sleep.c.bak，结果还是一样的报错

mysterywolf, 2021-07-06:
你用最新版的master

songzeceng, 2021-07-06:
> 你用最新版的master

最新版的又有新的错误：
1、移除k210的sleep.c，它说
```
scons: *** [build/packages/kendryte-sdk-v0.5.5/kendryte-standalone-sdk/lib/bsp/sleep.o] Source `packages/kendryte-sdk-v0.5.5/kendryte-standalone-sdk/lib/bsp/sleep.c' not found, needed by target `build/packages/kendryte-sdk-v0.5.5/kendryte-standalone-sdk/lib/bsp/sleep.o'.
```
于是我又恢复了k210的sleep.c，随即出现第二个错误

2、不移除sleep.c
```
LINK rtthread.elf
/home/rtlab/szc/riscv-none-embed-gcc-10.1.0-x64/bin/../lib/gcc/riscv-none-embed/10.1.0/../../../../riscv-none-embed/bin/ld: build/packages/kendryte-sdk-v0.5.5/kendryte-standalone-sdk/lib/bsp/sleep.o: in function `usleep':
sleep.c:(.text.usleep+0x0): multiple definition of `usleep'; build/kernel/components/libc/compilers/common/unistd.o:unistd.c:(.text.usleep+0x0): first defined here
/home/rtlab/szc/riscv-none-embed-gcc-10.1.0-x64/bin/../lib/gcc/riscv-none-embed/10.1.0/../../../../riscv-none-embed/bin/ld: build/packages/kendryte-sdk-v0.5.5/kendryte-standalone-sdk/lib/bsp/sleep.o: in function `msleep':
sleep.c:(.text.msleep+0x0): multiple definition of `msleep'; build/kernel/components/libc/compilers/common/delay.o:delay.c:(.text.msleep+0x0): first defined here
/home/rtlab/szc/riscv-none-embed-gcc-10.1.0-x64/bin/../lib/gcc/riscv-none-embed/10.1.0/../../../../riscv-none-embed/bin/ld: build/packages/kendryte-sdk-v0.5.5/kendryte-standalone-sdk/lib/bsp/sleep.o: in function `sleep':
sleep.c:(.text.sleep+0x0): multiple definition of `sleep'; build/kernel/components/libc/compilers/common/unistd.o:unistd.c:(.text.sleep+0x0): first defined here
collect2: error: ld returned 1 exit status
```
这些sleep()函数居然重复定义了

mysterywolf, 2021-07-06:
不是  我们有k210的软件包 刚才发的那个就是 用latest版本 最新版 已经帮你移除了 不需要你手动移除

songzeceng, 2021-07-06:
> 不是 我们有k210的软件包 刚才发的那个就是 用latest版本 最新版 已经帮你移除了 不需要你手动移除

不手动移除就是第二个错误：usleep()、msleep()、sleep()分别重复定义在了unistd.c、delay.c和unistd.c中，而且这是k210的包就是最新latest的

mysterywolf, 2021-07-06:
kendryte-sdk-v0.5.5 你选的不是latest版本 是0.5.5版本

songzeceng, 2021-07-06:
> kendryte-sdk-v0.5.5 你选的不是latest版本 是0.5.5版本

nice，原来是代码老了~

closed 2021-07-06 by songzeceng

mysterywolf, 2021-07-06:
我已经发布v0.5.7版本了 明天你应该就能看见了