RT-Thread/rt-thread #4869: RTthread erx 线程异常

shaoxiaoxi, 2021-07-11:
做了个简单的web服务器，结果在PC端用浏览器访问web服务器发现问题了，大概访问30多次 ，rtthread底层的erx线程就挂掉了。
在rt_mb_recv函数中rt_schedule();以后，似乎就再也回不来了。之后，客户端就再也ping不通stm32的服务端了
在发数据send的时候也遇到过 这样的问题。

erx线程如下：
void eth_rx_thread_entry(void* parameter)
{
        struct eth_device* device;

        while (1)
        {
                if (rt_mb_recv(&eth_rx_thread_mb, (rt_uint32_t*)&device, RT_WAITING_FOREVER) == RT_EOK)
                {
                        struct pbuf *p;

                        /* receive all of buffer */
                        while (1)
                        {
                                           if(device->eth_rx == RT_NULL) break;
                p = device->eth_rx(&(device->parent));
                if (p != RT_NULL)
                {
                    /* notify to upper layer */
                    if( device->netif->input(p, device->netif) != ERR_OK )
                    {
                        LWIP_DEBUGF(NETIF_DEBUG, ("ethernetif_input: Input error\n"));
                        pbuf_free(p);
                        p = NULL;
                    }
                }
                else break;
                        }
                }
        }
}

rt_err_t rt_mb_recv (rt_mailbox_t mb, rt_uint32_t* value, rt_int32_t timeout)
{
........
rt_schedule();
.........
}

BernardXiong, 2021-07-13:
内存越界了

shaoxiaoxi, 2021-07-13:
> 内存越界了

shaoxiaoxi, 2021-07-13:
现象如下：
1. list_mailbox显示erx的suspendlist为空，邮箱有条目  erxmb 0010 0016 0
2. erx 任务再也无法调度，具体现象参见https://club.rt-thread.org/ask/question/10467.html

分析：
1. rt_mb_recv 关全局中断在邮箱为空时挂起erx任务，将erx添加到suspend_thread列表中，开全局中断并触发任务切换。
2. rt_mb_send_wait 流程类似，检查erxmb的suspend_thread列表是否空，非空唤醒erx任务。
3. 经过步骤1， suspend_thread链表节点和erx的thread->tlist链表节点互相指向对方。
   步骤2 恢复suspend_thread, 并断开链表

疑问：
    1、是否存在一种场景？erx的tlist发生改变比如rt_list_remove(&(thread->tlist))，导致erxmb的suspend_thread意外置空。因此rt_mb_send_wait 流程里无法正常唤醒erx，进而导致erx任务一直处于挂起而无法得到调度。
    2、mailbox 代码保证suspend_thread的线程安全，但是并不保证thread->tlist不会改变 
    3、这个问题是否和内核调度有关？https://club.rt-thread.org/ask/question/10467.html所列问题其实只是因为触发条件严苛更难复现，并没有解决。

shaoxiaoxi, 2021-07-13:
> 内存越界了

感觉和内存越界没什么关系，更详细的分析如下：
现象如下：

list_mailbox显示erx的suspendlist为空，邮箱有条目 erxmb 0010 0016 0
erx 任务再也无法调度，具体现象参见https://club.rt-thread.org/ask/question/10467.html
分析：

rt_mb_recv 关全局中断在邮箱为空时挂起erx任务，将erx添加到suspend_thread列表中，开全局中断并触发任务切换。
rt_mb_send_wait 流程类似，检查erxmb的suspend_thread列表是否空，非空唤醒erx任务。
经过步骤1， suspend_thread链表节点和erx的thread->tlist链表节点互相指向对方。
步骤2 恢复suspend_thread, 并断开链表
疑问：
1、是否存在一种场景？erx的tlist发生改变比如rt_list_remove(&(thread->tlist))，导致erxmb的suspend_thread意外置空。因此rt_mb_send_wait 流程里无法正常唤醒erx，进而导致erx任务一直处于挂起而无法得到调度。
2、mailbox 代码保证suspend_thread的线程安全，但是并不保证thread->tlist不会改变
3、这个问题是否和内核调度有关？https://club.rt-thread.org/ask/question/10467.html所列问题其实只是因为触发条件严苛更难复现，并没有解决。

weiyf1003, 2021-08-12:
相同的问题，相同的现象。
erx线程和mb脱钩，list_mailbox下erxmb没有线程挂在suspend list上，mb->entry > 0，没人收。
但是etx是OK的，能抓到ARP，至今悬而未解。也考虑过内存越界问题，不过更可能是erxmb的list出了问题。

weiyf1003, 2021-08-12:
https://club.rt-thread.org/ask/question/432183.html
基本情况和论坛这个帖子是一致的
@shaoxiaoxi，看看是不是相同的问题？

weiyf1003, 2021-08-12:
最近又出现了另一个诡异的事情，LwIP 2.1.2 + iperf测试，2000-4000s会发生tcpip的mailbox异常
[image]

BernardXiong, 2021-08-13:
master中间一个版本会存在脱钩僵尸线程的问题，新的版本应该没这个问题。不知道你这个问题是否相关，如果已经是最新版本了，还是怀疑内存越界的可能性会更大些。

Anyway，这类问题都是需要仔细来分析的，没有好的办法。

weiyf1003, 2021-10-09:
要先理解前提条件：
1. systick如果配置了中断优先级那么会低于外设中断，也就是在systick中断里会被外设打断；
2. rt_tick_increase目前的release没加临界资源保护，里面thread->stat |= YIELD存在特定情况隐患；
3. 在下述相对极端的情况下：
    3.1 erx线程的rt_mb_recv中，如果mb entry = 0则erx挂起在suspend list里，此时erx thread stat = SUSPEND;
    3.2 rt_mb_recv接下来会开        
          /* enable interrupt */
          rt_hw_interrupt_enable(temp);
          /* re-schedule */
          rt_schedule();
    3.3 在enable interrupt后schedule前如果发生了systick中断，那么rt_tick_increase中thread->stat值是SUSPEND；
    3.4 因为 |= 并非原子操作汇编是3条语句，先把stat放到R0寄存器，然后去或，或完再赋值回来；
    3.5 当SUSPEND放入R0但是未赋值完成时，如果来了以太网接收中断，那么systick被以太网接收中断打断掉；
    3.6 以太网接收中断会发送mail调用rt_mb_send，rt_mb_send发现suspend list存在会把suspend list删掉，timer关闭，erx的stat置为READY并尝试唤醒erx；
    3.7 在rt_mb_send最后enable interrupt后回到rt_tick_increase中继续进行 |= 的后两步，此时内部R0是SUSPEND的，但是erx的stat已经被以太网中断更改了，并且suspend list被删除（这里是重点！！！），后赋值回来 erx线程在经过完整的systick中断后stat值是 SUSPEND | YIELD （0x0A）;
    3.8 再回到原有的环境，如果后续又来了以太网的接收中断，在rt_mb_send里if (!rt_list_isempty(&mb->parent.suspend_thread))会一直判假，产生erx线程假死的情况。根本原因是在3.6和3.7里suspend list已经删除，erx stat = 0x0A；

你们可以这样去debug：
1. 在rt_mb_recv函数的enable interrupt前，enable interrupt和schedule之间以及schedule后分别加变量去自增（x++,y++,z++）,理论上x == y == z发生问题后x会比y多1，也就是开了中断再也回不来了；
2. 把线程的stat打印出来，比如 static void show_stat(int argc, char **argv)入参线程名，FinSH里打印thread->stat；发生问题后stat会是0x0A；

最后在Master分支里，6月1号我看clock.c已经修复了这个BUG，不妨试一试将Master分支的clock.c替换到你们的工程。

@shaoxiaoxi 
@cychen88 
@BernardXiong

thewon86, 2021-11-10:
> 要先理解前提条件：
> 
> 1. systick如果配置了中断优先级那么会低于外设中断，也就是在systick中断里会被外设打断；
> 2. rt_tick_increase目前的release没加临界资源保护，里面thread->stat |= YIELD存在特定情况隐患；
> 3. 在下述相对极端的情况下：
>    3.1 erx线程的rt_mb_recv中，如果mb entry = 0则erx挂起在suspend list里，此时erx thread stat = SUSPEND;
>    3.2 rt_mb_recv接下来会开
>    /* enable interrupt _/
>    rt_hw_interrupt_enable(temp);
>    /_ re-schedule */
>    rt_schedule();
>    3.3 在enable interrupt后schedule前如果发生了systick中断，那么rt_tick_increase中thread->stat值是SUSPEND；
>    3.4 因为 |= 并非原子操作汇编是3条语句，先把stat放到R0寄存器，然后去或，或完再赋值回来；
>    3.5 当SUSPEND放入R0但是未赋值完成时，如果来了以太网接收中断，那么systick被以太网接收中断打断掉；
>    3.6 以太网接收中断会发送mail调用rt_mb_send，rt_mb_send发现suspend list存在会把suspend list删掉，timer关闭，erx的stat置为READY并尝试唤醒erx；
>    3.7 在rt_mb_send最后enable interrupt后回到rt_tick_increase中继续进行 |= 的后两步，此时内部R0是SUSPEND的，但是erx的stat已经被以太网中断更改了，并且suspend list被删除（这里是重点！！！），后赋值回来 erx线程在经过完整的systick中断后stat值是 SUSPEND | YIELD （0x0A）;
>    3.8 再回到原有的环境，如果后续又来了以太网的接收中断，在rt_mb_send里if (!rt_list_isempty(&mb->parent.suspend_thread))会一直判假，产生erx线程假死的情况。根本原因是在3.6和3.7里suspend list已经删除，erx stat = 0x0A；
> 
> 你们可以这样去debug：
> 
> 1. 在rt_mb_recv函数的enable interrupt前，enable interrupt和schedule之间以及schedule后分别加变量去自增（x++,y++,z++）,理论上x == y == z发生问题后x会比y多1，也就是开了中断再也回不来了（这里要判断mb是不是erx mailbox否则系统内部mailbox非常多定位会出问题，只有erx mailbox才去进行x++，y++，z++）；
> 2. 把线程的stat打印出来，比如 static void show_stat(int argc, char **argv)入参线程名，FinSH里打印thread->stat；发生问题后stat会是0x0A；
> 
> 最后在Master分支里，6月1号我看clock.c已经修复了这个BUG，不妨试一试将Master分支的clock.c替换到你们的工程。
> 
> @shaoxiaoxi @cychen88 @BernardXiong

回头看了很多遍，终于理解你说的了。这种情况也太巧了，恰恰在自己的时间片耗尽的时候，来了发给自己的消息。
既然知道是先 SUSPEND 后 YIELD 出问题的。可不可以先判断是否 SUSPEND 然后决定要不要进行 YIELD？

closed 2021-11-12 by shaoxiaoxi