RT-Thread/rt-thread #4884: ls1c stack overflow

woshinieao, 2021-07-23:
board  ls1c300b
os :     rtthread

error:
current sr: 0x1000fc00

 \ | /
- RT -     Thread Operating System
 / | \     4.0.4 build Jul 23 2021
 2006 - 2021 Copyright by rt-thread team
lwIP-1.4.1 initialized!
eth_inited!
warning: etx stack is close to end of stack address.
[W/TIME] Cannot find a RTC device to provide time!
File System initialzation failed!
warning: erx stack is close to end of stack address.
warning: etx stack is close to end of stack address.
thread:erx stack overflow

lwip  tx thread stack  is too small ， and kernel  start error ！

then ， i dissable the lwip， everything is ok ！！

mysterywolf, 2021-07-23:
尝试扩大一下tx线程的线程栈大小

mysterywolf, 2021-09-18:
你好可否提个pr？