RT-Thread/rt-thread #4927: rt-thread 是否支持运行外部程序

Rrooach, 2021-08-06:
hi，
我刚接触 rt-thread，也是刚接触RTOS。比较好奇rt-thread是否可以运行外部编译好的程序（比如使用rt-thread+qemu时在外部编译好可执行程序然后直接传到qemu中执行）。同时比较好奇rt-thread 是否支持直接执行syscall这样的方式对系统进行操作

BernardXiong, 2021-08-06:
请查看rt-smart的情况（rt-smart分支，或者说RT-Thread Smart）

Rrooach, 2021-08-07:
不好意思我并没有在rt-smart分支上看到什么有用的信息，请问一下能有更多的提示吗。
btw，我发现 rt-thread 内部有支持 syscall 调用，比较好奇 rt-thread 是否可以直接执行外部对 syscall 的一些调用，如果可以的话，应该怎么实现呢

BernardXiong, 2021-08-10:
感谢反馈，目前暂时可以参考这份[文档链接](https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-smart/rt-smart-quickstart/rt-smart-quickstart)

后期我们在document目录下加入更多。

jasonleecode, 2021-08-25:
要想实现在外部编译好程序放到rt-thread上运行，推荐rt-thread+wamr组合

majianjia, 2021-09-26:
@jasonleecode 
你说的是这个么？可以把外部编译的代码作为线程运行或者作为动态库提供api
https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/dlmodule/dlmodule

closed 2021-11-17 by mysterywolf