RT-Thread/rt-thread #4928: 使用 ARMLIBC fflush API 可能无法真正写入数据的问题

Tennn11, 2021-08-06:
[image]

如上述代码，当执行 libc 里面的 `fflush` API 时，底层会调用 `_sys_write` 函数，可是该函数只调用了 posix 的 `write` 函数，如果底层是 FatFS 时，此时数据并不会真正写入 flash 中。

这个问题我的考虑是不是在 `write` 函数下面加个 `fsync` 函数更好呢？

mysterywolf, 2021-08-06:
这个是fatfs的特性，write只是写到了缓存中，只有调用sync或者close时才会写到真的写到文件里。
这个行为模式和linux是一样的：https://blog.csdn.net/bandaoyu/article/details/83269012
当然你可以去linux用他的write函数再验证一下。一切以linux行为为准。
如果在write下面加sync，后果就是存储块设备磨损会比较大。

BernardXiong, 2021-08-11:
write只管write的，当有文件系统存在时，则调用底层的write实现去写入 ；并是说这里需要调用fsync的，如果是bsd socket是怎么办？

mysterywolf, 2021-11-17:
> write只管write的，当有文件系统存在时，则调用底层的write实现去写入 ；并是说这里需要调用fsync的，如果是bsd socket是怎么办？

这个和socket没有关系，这个是给fread fwrite函数用的，不是unix-like函数。
但是如果改成后面加个fsync，会导致sd卡的寿命收到影响，因为fwrite每一句都会实际写入到sd卡或者其他存储介质中。

mysterywolf, 2022-02-16:
https://github.com/ARMmbed/mbed-os/issues/1601

mysterywolf, 2022-08-16:
https://github.com/RT-Thread/rt-thread/pull/6286

closed 2022-08-16 by mysterywolf