RT-Thread/rt-thread #5000: RTT Studio 野火霸道板子切换系统版本出问题

loogg, 2021-09-01:
使用 RTT-Studio 基于开发板创建野火霸道工程，默认的系统版本是 4.0.3 ，切换版本到 latest 时出问题，rt-thread 文件夹少了很多文件

mysterywolf, 2021-09-01:
看看master有没有同样的问题，如果有就提个pr改了吧
我一般都是从master上用scons --dist-ide命令 dist一份 然后导入到studio里边，不愿意用studio自带的bsp包，脱节太严重。

loogg, 2021-09-02:
> 看看master有没有同样的问题，如果有就提个pr改了吧
> 我一般都是从master上用scons --dist-ide命令 dist一份 然后导入到studio里边，不愿意用studio自带的bsp包，studio的bsp脱节太严重。

loogg, 2021-09-02:
> > 看看master有没有同样的问题，如果有就提个pr改了吧
> > 我一般都是从master上用scons --dist-ide命令 dist一份 然后导入到studio里边，不愿意用studio自带的bsp包，studio的bsp脱节太严重。

目前 Studio 里 ST 的包都是通过 scons 生成的了，和以前不一样了

mysterywolf, 2021-09-22:
> 目前 Studio 里 ST 的包都是通过 scons 生成的了，和以前不一样了

不是的 master的bsp和studio的bsp还是脱节的 不是现生成的

loogg, 2021-09-23:
> > 目前 Studio 里 ST 的包都是通过 scons 生成的了，和以前不一样了
> 
> 不是的 master的bsp和studio的bsp还是脱节的 不是现生成的

不是啊，studio的bsp就是scons生成的，我还问过郭老师的

mysterywolf, 2021-09-23:
那个是不定期用scons --dist-ide人工生成的 不是rtstudio在创建的时候现生成的。

mysterywolf, 2021-10-12:
这个问题能提个pr修了嘛

closed 2021-12-17 by mysterywolf