RT-Thread/rt-thread #5059: 使用ulog组件时编译错误

shishirong, 2021-09-14:
芯片 STM32F103
工具链：KEIL MDK 
报错信息：
```c
../../../components/utilities/ulog/ulog_def.h(208): error: unknown type name 'size_t'
../../../components/utilities/ulog/ulog.c(119): error: use of undeclared identifier 'NULL'
```

在ulog_def.h中加上如下的头文件包含后编译通过：
```c
#include <rtdef.h>
#include <stddef.h>
```

Guozhanxin, 2021-09-14:
正在处理中，不久就修复。

parjing, 2021-09-15:
> 正在处理中，不久就修复。
不止有楼主提的问题，当内核使用标准库时，出现strnlen（rt_strnlen）找不到实体函数（链接不上）：
.\build\keil\Obj\rt-thread.axf: Error: L6218E: Undefined symbol strnlen (referred from ulog.o).

rt-thread/components/utilities/ulog/ulog.c(282): warning: implicit declaration of function 'localtime_r' is invalid in C99 [-Wimplicit-function-declaration]
            tm = localtime_r(&t, &tm_tmp);
                 ^
rt-thread/components/utilities/ulog/ulog.c(282): warning: incompatible integer to pointer conversion assigning to 'struct tm *' from 'int' [-Wint-conversion]
            tm = localtime_r(&t, &tm_tmp);
               ^ ~~~~~~~~~~~~~~~~~~~~~~~~
rt-thread/components/utilities/ulog/ulog.c(1253): warning: extra tokens at end of #endif directive [-Wextra-tokens]
#endif defined(RT_USING_FINSH)
       ^
还有相关告警也希望能够修复下

mysterywolf, 2021-09-28:
https://club.rt-thread.org/ask/article/3081.html

mysterywolf, 2021-09-28:
https://club.rt-thread.org/ask/question/433010.html

mysterywolf, 2021-09-28:
@Guozhanxin 郭老师 这些都是缺<stdint.h> <stddef.h>的 麻烦找人处理一下吧 俺就不管了😁

closed 2021-10-09 by mysterywolf