RT-Thread/rt-thread #5060: risc-v  context switch  interrupt  error

woshinieao, 2021-09-14:
rtthread  risc-v 中的context_gcc.S  任务切换的时候 ，过早的打开了中断 ，存在隐患。

代码如下：
`
 rt_hw_context_switch_exit:

    /* resw ra to mepc */
    LOAD a0,   0 * REGBYTES(sp)
    csrw mepc, a0

    LOAD x1,   1 * REGBYTES(sp)

    **li    t0, 0x00007800
    csrw  mstatus, t0
    LOAD a0,   2 * REGBYTES(sp)
    csrs mstatus, a0**

    LOAD x4,   4 * REGBYTES(sp)
    LOAD x5,   5 * REGBYTES(sp)
    LOAD x6,   6 * REGBYTES(sp)
    LOAD x7,   7 * REGBYTES(sp)
    LOAD x8,   8 * REGBYTES(sp)
    LOAD x9,   9 * REGBYTES(sp)
    LOAD x10, 10 * REGBYTES(sp)
    LOAD x11, 11 * REGBYTES(sp)
    LOAD x12, 12 * REGBYTES(sp)
    LOAD x13, 13 * REGBYTES(sp)
    LOAD x14, 14 * REGBYTES(sp)
    LOAD x15, 15 * REGBYTES(sp)
    LOAD x16, 16 * REGBYTES(sp)
    LOAD x17, 17 * REGBYTES(sp)
    LOAD x18, 18 * REGBYTES(sp)
    LOAD x19, 19 * REGBYTES(sp)
    LOAD x20, 20 * REGBYTES(sp)
    LOAD x21, 21 * REGBYTES(sp)
    LOAD x22, 22 * REGBYTES(sp)
    LOAD x23, 23 * REGBYTES(sp)
    LOAD x24, 24 * REGBYTES(sp)
    LOAD x25, 25 * REGBYTES(sp)
    LOAD x26, 26 * REGBYTES(sp)
    LOAD x27, 27 * REGBYTES(sp)
    LOAD x28, 28 * REGBYTES(sp)
    LOAD x29, 29 * REGBYTES(sp)
    LOAD x30, 30 * REGBYTES(sp)
    LOAD x31, 31 * REGBYTES(sp)

    addi sp,  sp, 32 * REGBYTES
    mret

`

进入任务切换后半段程序 ，首先是打开了了中断，然后才开始加载程序栈中相关的寄存器值，如果在这个过程中发生中断事件，程序后续如何走 ？

chenyingchun0312, 2021-09-27:
在任务切换过程中，确实需要关闭全局中断，我最近刚好也在看riscv的任务切换代码，一起交流下，我没有特别看到你说的，开启中断的位置？

如下链接是我最近分析的一些注释，不完全正确，可以一起交流下
https://github.com/chenyingchun0312/rt-thread/blob/riscv/libcpu/risc-v/common/context_gcc.S

chenyingchun0312, 2021-10-08:
@woshinieao Hi,我最近又研究了下任务切换代码，应该不存在你说的问题，我的分析如下：

[image]

chenyingchun0312, 2021-10-15:
如果你说的下半部分，装载CPU之前，打开了中断是下面截图这部分的话，我觉得是不是你理解错了？这里应该是没有打开中断的，即mstatus.mie此时还是0（ 上面的代码有关闭全局中断操作),并未变成1

[image]

chenyingchun0312, 2021-10-15:
@woshinieao 可以再看一下，我上面的理解，是否有问题

mysterywolf, 2022-01-23:
@woshinieao 是否还有疑问 如果没有疑问的话我就关闭issue了

closed 2022-01-23 by mysterywolf