RT-Thread/rt-thread #5061: sem release 没有引起真正的调度事件发生

woshinieao, 2021-09-14:
rtthread 中相同优先级 sem relase 并不会真正的发生线程的调度。

代码如下：
`
void test1_thread_entry(void* parameter)
{
	static rt_uint16_t count = 0;
	int rev_len = 0;
	rt_uint8_t len,*rx_buff;
	
	while(1)
	{
			rt_kprintf("1\n");
			rt_thread_delay(2000);
			rt_sem_release(&sem1);
			rt_kprintf("1......\n");
	}

}

void test2_thread_entry(void* parameter)
{
	rt_uint32_t sum = 0;

	while(1)
	{

		rt_sem_take(&sem1, RT_WAITING_FOREVER);
		{	
			rt_kprintf("2\n");
		}
	}
    
}

`

输出结果如下：
`
 \ | /
- RT -     Thread Operating System
 / | \     3.0.1 build Dec 29 2020
 2006 - 2017 Copyright by rt-thread team
1
[16:45:40.866] 1......
1
2
[16:45:42.867] 1......
1
2
[16:45:44.867] 1......
1
2
[16:45:46.867] 1......
1
2

`

线程启动后 ，第一次打印的1 和1......  中间并没有出现2  ， 所以sem  release 并没有引起线程的调度。知道delay才真正发生了调度事件

Guozhanxin, 2021-09-14:
初值是多少呢？可以把完整的例子贴上来。

sidaiji, 2021-09-15:
> 初值是多少呢？可以把完整的例子贴上来。

你好，信号量初值为0

hzcx998, 2021-10-14:
貌似你这个版本有点低，可以试试比较新的版本，看看是否仍然有这个问题，如果是就是一直没有修复这个问题，如果不是，那么就是已经修复了，就可以看看哪个版本修复了这个问题。

grissiom, 2022-03-04:
相同优先级为啥要调度呢？可以等当前优先级的跑完在跑 ready 的，反正都一样

closed 2022-06-17 by mysterywolf