RT-Thread/rt-thread #5073: 串口使用POSIX接口 ioctrl 无法设置底层参数，不会调用底层config 由于宏定义的数值为3

chunyexixiaoyu, 2021-09-17:
[image]

[image]

mysterywolf, 2021-09-17:
也就是posix标准的数值和rt这边的定义数值重复了？

mysterywolf, 2021-09-17:
你这个使用方法有些问题，ioctl是posix层级的，不要使用rt设备层级的宏。如果你想要这么调用，请使用rt_device_control
很显然在unix编程中，ioctl不可能会调用RT_DEVICE_CONTROL_CONFIG这个宏。

chunyexixiaoyu, 2021-09-18:
> 你这个使用方法有些问题，ioctl是posix层级的，不要使用rt设备层级的宏。如果你想要这么调用，请使用rt_device_control
> 很显然在unix编程中，ioctl不可能会调用RT_DEVICE_CONTROL_CONFIG这个宏。
老师，那我想用ioctl控制串口配置串口应该怎么做呢？

chunyexixiaoyu, 2021-09-18:
> 你这个使用方法有些问题，ioctl是posix层级的，不要使用rt设备层级的宏。如果你想要这么调用，请使用rt_device_control
> 很显然在unix编程中，ioctl不可能会调用RT_DEVICE_CONTROL_CONFIG这个宏。
@mysterywolf   老师，如果我想用ioctrl控制配置底下串口应该怎么做呢？

mysterywolf, 2021-09-18:
@Guozhanxin  郭老师看看这个问题是否需要修改？

mysterywolf, 2021-09-21:
[image]
此处确认为一个bug @Guozhanxin

chunyexixiaoyu, 2021-09-22:
@mysterywolf 你说的也有道理 RT_DEVICE_CTRL_CONFIG 这个是rtt操作系统特有的宏定义，posix接口的设计是否中间做一个转换印射到rtt的设备宏

mysterywolf, 2022-03-05:
https://club.rt-thread.org/ask/question/430135.html 跟踪到社区论坛也报告了一起

mysterywolf, 2022-05-31:
https://github.com/RT-Thread/rt-thread/pull/5769 这个PR解决了类似这种问题 但是没有处理到串口

CXSforHPU, 2026-10-13:
当前问题并未解决，该pr  https://github.com/RT-Thread/rt-thread/pull/5769 ,虽然是细化的。但其只覆盖了部分。且该pr并未影响串口的ctrl值。并且当前问题仅存在于dfs_v1中，解决问题需要继续细化rt-thread\components\drivers\include\drivers中的ctrl

CXSforHPU, 2026-10-13:
我希望能够接手该issue，并推荐其关闭