RT-Thread/rt-thread #5110: Update STM32 HAL drivers to support latest MCU

majianjia, 2021-09-28:
Some STM32 HAL drivers are too old (Some over 2 years old) now that have difficulty supporting the latest MCU. Also, there are many bugs fixed from ST. 
For example, the STM32H7 HAL is v1.7 which doesn't support STM32H7A/H7B series.

Any plan to do a good upgrade on all STM32's HAL?
This kind of work is better to do little by little than after a long time until too much gap to fill. 
I would suggest catching up with STM32's HAL updates after a major work to avoid another major.

mysterywolf, 2021-09-28:
HAL库刚更新完😅

majianjia, 2021-09-28:
> HAL库刚更新完😅 #4985

不够啊，还是差了两年

mysterywolf, 2021-09-28:
啊？ 那这是咋更新的。。。

mysterywolf, 2021-09-28:
@MrYangMouren 你好 请确认一下stm32 hal库是否提升到了最新版本😂

majianjia, 2021-09-29:
G0, G4, H7, L4, WB, MP 等跟最新版差了1到2个版本号。

Guozhanxin, 2021-09-29:
有最新版hal库的下载链接吗？

majianjia, 2021-09-29:
@Guozhanxin 
下载一个最新的STM32CubeMX 然后在里面就可以下载所有HAL的版本了。下载完后磁盘里面就可以看到相应的版本HAL。

mysterywolf, 2021-10-02:
https://github.com/RT-Thread/rt-thread/issues/4794#issue-919742766 这里边的仓库时最新的

liukangcc, 2021-10-09:
https://github.com/RT-Thread/rt-thread/pull/5129