RT-Thread/rt-thread #5307: [SDIO] Multi-host supporting

wosayttn, 2021-11-26:
Hi,

最近在研究 RT_Device_Class_SDIO 的對接，這是個好物; 但在 Multi-host 下，似乎 auto-mounting 是不支持的?  它目前僅支持單張 SD 卡上的所有 FAT partition 掛載; 但似乎不支持 多張卡多區 的掛載?
EX: 
Now: sd0, sd1, ......., sdN.  
Wish: sda1, sda2, ......, sdaN; sdb1, sdb2, ....,sdbN.

[Mounting for Signle](https://github.com/RT-Thread/rt-thread/blob/master/components/drivers/sdio/block_dev.c#L402 )

BernardXiong, 2021-11-28:
欢迎提建议。

多SD/MMC，当前框架是支持的，不过好像是分区不支持？

Guozhanxin, 2021-11-29:
这里有个pr，看看是不是相关的？https://github.com/RT-Thread/rt-thread/pull/5062

wosayttn, 2021-11-29:
@BernardXiong 目前不支持的。
@Guozhanxin 不像我的wish。

我搞了 一版支持這個功能 且 提供一BUGFIX，請大師們 comment and approve to merge.
[Implemented](https://github.com/RT-Thread/rt-thread/pull/5312)

closed 2021-12-28 by wosayttn